GeoNet/help #5: WFS modificationtime filter treating 'greater than' ( > ) as 'greater than or equal to' ( >= )

adamsp, 2015-12-27:
I'm retrieving latest data from the Web Feature Service. I'm retrieving it in pages with queries similar to the following:

http://wfs.geonet.org.nz/geonet/ows?service=WFS&version=1.0.0&request=GetFeature&typeName=geonet:quake_search_v1&outputFormat=json&sortBy=modificationtime&cql_filter=modificationtime%3E2015-12-26T18:43:52.967Z%20AND%20eventtype=%27earthquake%27&maxFeatures=50

This gives me chunks of 50 items at a time. When I retrieve a page I store the last `modificationtime` and use that in my query for the next page.

The problem I'm facing is that `modificationtime>2015-12-26T18:43:52.967Z` (for any date) is returning as if I was supplying a  `>=` operator. This means that whenever I check for newly modified events, my "last seen" event is always included in the returned data.

You can reproduce this by running the above query (drop the `maxFeatures` filter), taking the last event's `modificationtime` and replacing the one in the query with that one. The expectation is that we get 0 results; instead we get 1.

This is trivial to workaround client-side (it just runs through my logic for handling a modification to an existing event), however it does appear to be a bug so I figured I'd report it anyway. I had a quick glance through your public repos and couldn't find anything for this service or I'd endeavour to fix it myself.

Cheers!

kfenaugh, 2016-01-04:
Hi Adam,

This is puzzling. I tried another workaround, which was to add a zero to the end of the modification timestamp. ie.

http://wfs.geonet.org.nz/geonet/ows?service=WFS&version=1.0.0&request=GetFeature&typeName=geonet:quake_search_v1&outputFormat=json&sortBy=modificationtime&cql_filter=modificationtime%3E2015-12-26T18:43:52.9670Z%20AND%20eventtype=%27earthquake%27&maxFeatures=50

This correctly left out the errant entry.

adamsp, 2016-01-05:
Excellent! I've tested this and it works for me - if I supply the second-to-last event's updated timestamp, I get the most recent event only, as expected. This is much cleaner as I can simply modify my date formatter used when constructing the request (along with a comment as to why we're deviating from ISO8601 for this one specific case).

Thanks for your help!

closed 2016-01-05 by adamsp