GeoNet/help #6: FDSN event query returning 204

calum-chamberlain, 2016-02-22:
Using [obspy.clients.fdsn.Client.get_events](https://github.com/obspy/obspy/blob/master/obspy/clients/fdsn/client.py#L282) with GeoNet attempts to query the GeoNet FSDN site with URLs like this:
http://service.geonet.org.nz/fdsnws/event/1/query?eventid=2016p008122.  These generate a 204 error message:
`FDSNException: No data available for request.
`
Which arises from [here](https://github.com/obspy/obspy/blob/master/obspy/clients/fdsn/client.py#L1327).  

The same error is generated for other queries, I have tried start and end time flags, and locations.

Currently for known event id's my work around is:

    from obspy.clients.fdsn import Client
    
    publicIDs=['2016p008122', '2016p008353', '2016p008155', '2016p008194']
    client = Client("GEONET")
    
    catalog = Catalog()
    for publicID in publicIDs:
        data_stream = client._download('http://quakeml.geonet.org.nz/' +
                                       'quakeml/1.2/' + publicID)
        data_stream.seek(0, 0)
        catalog += read_events(data_stream, format="quakeml")
        data_stream.close()

Which is not ideal and only works for known eventID numbers.

Not sure of a fix!

kfenaugh, 2016-02-23:
Calum,

As documented on this page http://info.geonet.org.nz/x/CQSD we have not implemented the 'event' service at this time. Thanks for the interest in having it as part of the suite. I'll look into where it could fit into our data services setup.

Kevin

calum-chamberlain, 2016-02-23:
Thanks for that Kevin, sorry I hadn't seen that.  It would be a really useful addition and if there is anything I can do to help I would be happy to.

closed 2016-05-26 by kfenaugh