GeoNet/help #7: Velocity model questions

nbalfour, 2016-04-26:
These questions are from a student at VUW regarding the velocity model we use with NonLinLoc. 

Is the velocity model used my NonLinLoc a true 3D model in that, it was not a 2D model projected in 3D space? 

If it is a true 3D velocity model in a NonLinLoc friendly format, is that velocity model available for use to others, say me (VUW student)? 

If so, where would i be able to obtain a copy? 

@salichon @kfenaugh

rumachan, 2016-04-26:
The model is available on ftp.geonet.org,nz/seiscomp/nll/3drx Not sure who put it there.

Of course it a real 3D model! Donna Eberhart-Phillips work. I'm sure someone has the reference.

salichon, 2016-04-26:
hi ,
 it s a SRL 2010 vol81 Nr 6 By Donna Eberhardt-Philips  et al doi 10.1785

salichon, 2016-04-26:
i guess the 3D files are available too using the online material

salichon, 2016-04-26:
link is here http://dx.doi.org/10.1785/gssrl.81.6.992

nbalfour, 2016-04-26:
Thanks for your help @rumachan and @salichon

closed 2016-04-26 by nbalfour