GeoNet/help #8: Volcano Auckland internal map - extend depth to 100 km

ndeligne, 2016-06-21:
The volcano monitoring team would like the internally used Auckland earthquake map (http://images.geonet.org.nz/maps/earthquakes-auckland.jpg) to extend the depth of earthquakes displayed on the map to 100 km (minuted at https://wiki.geonet.org.nz/display/volcano/20160620#id-20160620-Generalbusiness). The depth should remain 50 km for all other similar maps. Everything else about the Auckland map should remain the same (map, colors, symbols, etc.). The rationale behind this is that DEVORA and other research fairly convincingly show that the magma source region for Auckland is 90-100 km depth, and we would like the map to extend to the magma source region. 50 km captures the magma source region for other NZ volcanoes.

gclitheroe, 2016-06-21:
We also have these maps http://www.geonet.org.nz/volcano/info/aucklandvolcanicfield/quakes

1. are they worth retaining?
2. if so is it possible for someone to give us bounds (complex polygons are ok) and depth limits for them?
3. are these sorts of displays worth considering for the interactive viewer @crleblanc demonstrated?

ndeligne, 2016-06-21:
Hi Geoff,

These maps are worth retaining, I'll add a link to it on the DEVORA 
website. However, should extend the source region to 100 km.

 I'll get you a polygon within the next week - what format do you want it 
in? Shapefile, kmz file, polygon points?

I'm not sure what @crleblanc demonstrated? What did I miss?

Cheers,
Natalia

_____________________________________________
Dr. Natalia Irma Deligne
Volcanic Hazard and Risk Modeller
GNS Science - Te Pu Ao
PO Box 30368
Lower Hutt 5040
New Zealand

Email: N.Deligne@gns.cri.nz
Tel: +64 4 570 4129 / Fax: +64 4 570 4600

<author@noreply.github.com>
Date:   22/06/2016 09:32
depth to 100 km (#8)

http://www.geonet.org.nz/volcano/info/aucklandvolcanicfield/quakes

Notice: This email and any attachments are confidential.
If received in error please destroy and immediately notify us.
Do not copy or disclose the contents.

gclitheroe, 2016-06-21:
polygon points is fine. 

Do you have them for all the volcanic regions?  It's not urgent for the others but this is something we should consider improving for any web site redevelopment.

ndeligne, 2016-06-21:
I personally don't have them for other volcanic regions. For Auckland I 
will use polygon from a peer-reviewed paper that came out last year that 
provides a boundary. For other volcanoes we should probably look to QMAP? 
I will raise with the group at the next monitoring meeting.

_____________________________________________
Dr. Natalia Irma Deligne
Volcanic Hazard and Risk Modeller
GNS Science - Te Pu Ao
PO Box 30368
Lower Hutt 5040
New Zealand

Email: N.Deligne@gns.cri.nz
Tel: +64 4 570 4129 / Fax: +64 4 570 4600

Date:   22/06/2016 09:40
depth to 100 km (#8)

gclitheroe, 2016-06-21:
sounds great, thanks a lot.

rumachan, 2016-06-21:
We have two products (regional volcano area epicentre maps) that are fairly similar but have been developed for different target markets - a version used by the volcano monitoring team and one intended for public consumption. Those products were developed and are maintained by different people/teams within GeoNet The ideal situation would be to have a single product that, as much as possible, catered for both. This would require some compromise, especially on the part of the volcano monitoring team, but surely maintaining one product is better than two.

The internal team map shows some depth information and time-magnitude information that that team has gotten use to having, so a combined product would have to have that, in some way.  

The public map doesn't have a distinction between automatic and confirmed/manual locations, which I think is good. That distinction is a hangover from CUSP days and is no longer necessary. The public map also offers nice zooming and click-on-symbol to get more information. Neither of which we get in the internal map.

I'd like to see development of a unified product with as many of the requirements as possible, rather than a short-term tweaking of existing products.

rumachan, 2016-06-21:
Natalia,
I wouldn't use a Auckland Volcanic Centre polygon. The purpose of these maps is to show quakes in and near these volcanoes. A AVF polygon is too restrictive.

Good luck getting monitoring team agreement on map boundaries. When we decide how this is progressing (see my earlier comment) one or two making an executive decision is the only way to avoid wasting substantial time.

ndeligne, 2016-06-21:
Sounds good - I was going to supply to polygon and suggest a 10 - 20 km 
buffer around it to get more of the regional context. I'll get the 
conversation going about map extents for other regions. Are we tied to the 
Google Earth background map? Currently both available map backgrounds 
available on the public webpage don't show the level of topography that is 
helpful for monitoring team.

_____________________________________________
Dr. Natalia Irma Deligne
Volcanic Hazard and Risk Modeller
GNS Science - Te Pu Ao
PO Box 30368
Lower Hutt 5040
New Zealand

Email: N.Deligne@gns.cri.nz
Tel: +64 4 570 4129 / Fax: +64 4 570 4600

Date:   22/06/2016 09:58
depth to 100 km (#8)

ozym, 2016-06-21:
These are old maps and are on the list of update or throw out. Happy to hand over ...

For reference the rectangular bounds (needed for the type of layout) are as follows, given in NZMG:
* Tongariro

    1797000, 1852000, 5618000, 5687000
* Taupo

    1820000, 1884000, 5678000, 5731000
* Auckland

     1728000, 1797000, 5886000, 5955000
* Bay of Plenty

     1874000, 1978000, 5719000, 5879000
* Taranaki

     1642000, 1739000, 5593000, 5706000

rumachan, 2016-06-21:
The background on the public maps starts by showing GE, but as you zoom in you get a full topo map, which is more than enough detail for the monitoring team.

ndeligne, 2016-06-21:
I'll discuss with the team. Thanks!

_____________________________________________
Dr. Natalia Irma Deligne
Volcanic Hazard and Risk Modeller
GNS Science - Te Pu Ao
PO Box 30368
Lower Hutt 5040
New Zealand

Email: N.Deligne@gns.cri.nz
Tel: +64 4 570 4129 / Fax: +64 4 570 4600

Date:   22/06/2016 10:28
depth to 100 km (#8)

ndeligne, 2016-06-30:
Update: Volcanology ppl will be meeting on Monday to go over what we would like to have in map should they be changed to be used by use and the public. In the meantime, can the Auckland region depth be extended to 100 km? The revised maps will likely take some time, and we feel vulnerable not having the Auckland epicenter map not cover the inferred magma source region.

ozym, 2016-08-01:
band-aid applied

ndeligne, 2016-08-01:
thanks Mark!

closed 2016-10-10 by ozym