GeoNet/help #9: Gun Ridge (GUNR) cGPS

pdenys162, 2016-07-08:
The cGPS site GUNR (West Coast) appears to be located on a landside.  The site is unstable and the velocity is not representative of the region e.g. MTPR. (The vertical velocity is >-7 mm/yr).    Hence it is not a good site from a science (or anything else) perspective.   I would like to suggest that the site is moved to a location that would be more beneficial. 

Time Series and horizontal plots in attached files:

[GUNR.pdf](https://github.com/GeoNet/help/files/353372/GUNR.pdf)
[GUNR_hz.pdf](https://github.com/GeoNet/help/files/353373/GUNR_hz.pdf)

Some time ago, we did a recce to find sites to extend the SAGENZ profile north and south of the existing east-west profile (QUAR-MTJO) with the view of determining if the vertical velocities are increasing/decreasing north or south. We visited four sites,  which could be potential sites. 
  
Name of Conservation Area	Legal Status	Area
Architect Creek  H36 6331 or 6230 	Westland National Park
Cook River         H36 GS 6337 	        Westland National Park/Stewardship land
Mt Hermann      G36 GS4320-4420 	Stewardship land 
Bannock            G36 GS4122 	                Stewardship land

 
Unfortunately the grid references (grid square) are in terms of NZMG.    However, these sites have radio comms to either quarry (QUAR) or another of the SAGENZ cGPS site. 
 
I would propose that we remove GUNR and re-establish it in, say, the area identified near the cook river. I was wondering if this would be worthwhile investigating.   We would be happy to be involved. 

Paul Denys
School of Surveying

elidana, 2016-07-11:
Hi @pdenys162 ,

many thanks for taking care of our GUNR cGPS site, this is really appreciated.

We were aware of the problem since the end of 2014, and if you look at the site description for GUNR on our Delta database, you can find this warning:
> ON LARGE LANDSLIDE. Analysis of the data shows GUNR is moving atypically for regional tectonics and compared to nearby stations (eg MTPR). Affected by large landslide (also scarp visible to North). 

Our time series show the same problem you pointed out with your processing, as can be seen by comparing GUNR vertical to MTPR (Mt. Prince), located ~8 km NW of GUNR:
[image]

Unfortunately we were (and currently are) unable to relocate the site due to budget limitations, so we decided to leave the site operating, also because our scientists from the Landscape Department have some interest in monitoring the landslide.  

We really appreciate your offer to help and hopefully we will be able to relocate or build a new site in the future. In that case, we will surely get in touch with you, and your proposed sites will be our starting point for any site selection.

Kind regards,

Elisabetta

closed 2016-09-08 by elidana

elidana, 2018-05-15:
The daily position time series and long term velocity for GUNR (Gunn Ridge) site have been reviewed. 
The map below shows  the velocity field in the ITRF08 reference frame (horizontal velocities in purple, vertical velocities in blue).
[image]

We can confirm that the site is affected by local instabilities. For that reason the equipment (receiver and antenna) at the site have been removed and the station is considered as "closed".
The monument and mark have been left in place and available for future reoccupation.