GeoNet/help #10: Quakesearch maps not loading

calum-chamberlain, 2016-07-12:
Hey guys, not sure whether you are aware (assume you are, but thought seeing as no one had posted here yet I would), but quakesearch maps are not loading the background tiles, issue comes from MapQuest changing how you can access their database by the looks of it.

http://devblog.mapquest.com/2016/06/15/modernization-of-mapquest-results-in-changes-to-open-tile-access/

carolinelittle, 2016-07-12:
Thanks for letting us know Calum, we'll  get onto it.

carolinelittle, 2016-07-12:
@bpeng can you have a look into this?

carolinelittle, 2016-07-12:
Hi @calum-chamberlain, unfortunately it's not as simple a job as I thought, and there are other workwwork we'll need to do for Quakesearch. The open street map layer is still working though.

bpeng, 2016-07-12:
This affects all GeoNet web maps in fact, as discussed in https://github.com/GeoNet/tickets/issues/739. We'll need select another map server for satellite image tiles to replace MapQuest. For now the easiest solution would be to use the OSM Map as default layer.

closed 2016-09-07 by gclitheroe