GeoNet/help #11: Broken URL's - Network Map Images

michaelmurfy, 2016-08-14:
Just a simple one for you.

On the Seismic station information (eg: [this link](https://magma.geonet.org.nz/delta/app?service=external/ViewSeismicStation&sp=SNELS)) when clicking on the image it redirects to an image served via HTTP - I can see the server only responds to HTTPS meaning it just times out. This may also happen on other images pointing towards gallery.geonet.org.nz where it doesn't link to the HTTPS element. Maybe a sed is in order.

michaelmurfy, 2016-08-14:
Furthermore to this issue there appears to be broken images too (for example: [here](https://magma.geonet.org.nz/delta/app?service=external/ViewSeismicStation&sp=SMRNZ)) - just generating a 404.

carolinelittle, 2016-08-14:
Hi Michael, thanks for this. 

@dc735 can you have a look into this?

quiffman, 2016-08-15:
Hi Michael,

If fixing this was as easy as a quick `sed` or `awk`, rest assured, we would be all over it like a cheap suit!

Unfortunately, the software backing DELTA and its supporting imagery from gallery, are both a little more complex than that and due to their age and security concerns, we do not intend to fix this.

One of the things we are working hard towards is separating these components out in to more accessible streamlined projects, where something like a quick `sed` might actually be the fix this problem in the future.

As you rightly point out, the images that are linked from DELTA are actually available on gallery, if you modify the address to point at `https://gallery.geonet.org.nz/...`
And for the missing `SI` (South Island) network sitemaps, we do have a new sitemap image service available at (for example) http://images.geonet.org.nz/sitemaps/mrnz-si-location.gif

Thanks for the useful bug-report.
Possibly not the answer you were looking for, but hopefully other users will also find this information useful.
I'm going to close this and (for future reference) label `wontfix`.

closed 2016-08-15 by quiffman