GeoNet/help #12: CWB server down?

gclitheroe, 2016-08-25:
It looks like the CWB server is down.  As reported here https://github.com/GeoNet/GeoNetCWBQuery/issues/8

I can replicate the issue.

Please could you take a look?

gclitheroe, 2016-08-25:
```
java -jar GeoNetCWBQuery-4.2.0-bin.jar -lsc
Using default logging configuration.
line='-b' '2016/08/25 19:40:16.657' '-d' '300.0' '-lsc'
:
java.net.SocketException: Connection reset list channels
null
```

ozym, 2016-08-25:
Restarted the QueryServer process.

```
Using default logging configuration.
line='-b' '2016/08/25 21:46:12.988' '-d' '300.0' '-lsc'
:
There are 4825 channels
NZABAZ EHE10 #days=1    16_238
NZABAZ EHN10 #days=1    16_238
NZABAZ EHZ10 #days=1    16_238
NZABAZ LCQ01 #days=1    16_238
NZABAZ VEC01 #days=1    16_238
NZABAZ VEM01 #days=1    16_238
...

```

closed 2016-08-25 by ozym