GeoNet/help #13: Felt Report - Maps API key not activated

rachel-primrose, 2016-09-01:
https://felt.geonet.org.nz/

For any address.

Request:

```
Request URL:https://maps.googleapis.com/maps/api/place/js/AutocompletionService.GetPredictions?1sOrewa%2C%20Auckland&4sen-US&7scountry%3Anz&9sgeocode&15e3&key=AIzaSyBjYAoimxcNDMB4sDveVXAkNQxkgZjUoMc&callback=_xdc_._omyhkn&token=109638
Request Method:GET
Status Code:200 
Remote Address:[2607:f8b0:4005:801::200a]:443
```

Response:
```
alt-svc:quic=":443"; ma=2592000; v="36,35,34,33,32,31,30"
alternate-protocol:443:quic
cache-control:no-cache, must-revalidate
content-disposition:attachment
content-encoding:gzip
content-length:270
content-type:text/javascript; charset=UTF-8
date:Thu, 01 Sep 2016 17:54:22 GMT
expires:Fri, 01 Jan 1990 00:00:00 GMT
pragma:no-cache
server:pablo
status:200
x-frame-options:SAMEORIGIN
x-xss-protection:1; mode=block

/**/_xdc_._cmi34d && _xdc_._cmi34d( [3,null,null,"This API project is not authorized to use this API. Please ensure this API is activated in the Google Developers Console: https://console.developers.google.com/apis/api/places_backend?project=_ For more information on authentication and Google Maps Javascript API services please see: https://developers.google.com/maps/documentation/javascript/get-api-key"] )
```

quiffman, 2016-09-01:
Thanks for the report @rachel-primrose.

It looks like we've hit the limit on the Google Places API web service.
We are working on getting this sorted ASAP.

quiffman, 2016-09-01:
This should be fixed now.

Thanks again for the report @rachel-primrose!

closed 2016-09-01 by quiffman

rachel-primrose, 2016-09-01:
Thanks for the quick turnaround @quiffman! :+1: