GeoNet/help #15: Recent change to WFS has broken filtering

adamsp, 2016-09-05:
Hey folks,

I've got an application that retrieves earthquake events from the Web Feature Service. [According to the mailing list this was updated recently](http://lists.geonet.org.nz/pipermail/updates/2016-August/000003.html). Unfortunately, this update has broken my queries.

My queries look like this, and were working up until around 11 days ago:

> http://wfs.geonet.org.nz/geonet/ows?service=WFS&version=1.0.0&request=GetFeature&typeName=geonet:quake_search_v1&outputFormat=json&sortBy=modificationtime&cql_filter=modificationtime%3E2016-03-27T19:34:23.5650Z%20AND%20eventtype=%27earthquake%27&maxFeatures=50

And the response I now get has this body:

> found additional query parameters

I'm doing this specifically so that I can retrieve a paged delta of events since last time the client queried for events. This sorts by modification time so that I get oldest first, and everything since the last time I queried - so when I've finished with one page, I just query for the next page by asking for `modificationtime>[most recent event on previous page]`.

Now, the sorting, filtering and paging I can probably get by without, but the _delta_ functionality is pretty important to me. So if I pare that query back to the following:

> http://wfs.geonet.org.nz/geonet/ows?service=WFS&version=1.0.0&request=GetFeature&typeName=geonet:quake_search_v1&outputFormat=json&cql_filter=modificationtime%3E2016-03-27T19:34:23.5650Z

I get this:

> Invalid time format 2016-03-27T19:34:23.5650Z

Note the additional 0 on the milliseconds, [due to this issue](https://github.com/GeoNet/help/issues/5) - removing it doesn't help. I've also tried supplying a simple `long`, it doesn't like that either.

What time format can I use now? :smile:

bpeng, 2016-09-05:
Hi @adamsp 

Thank you for your feedback, this is a simple web service which provide basic WFS functionality, query examples can be found [here](http://wfs.geonet.org.nz/), it is not carefully tested.

As a workaround for your query, ```sortby``` is not currently supported, time format ```2016-03-27T19:34:23``` is OK, and the single quote in ```eventtype=`earthquake```` is not needed, try this:
```
http://wfs.geonet.org.nz/geonet/ows?service=WFS&version=1.0.0&request=GetFeature&typeName=geonet:quake_search_v1&outputFormat=json&cql_filter=modificationtime>2016-03-27T19:34:23+AND+eventtype=earthquake&maxFeatures=50
```

We will fix this issue in the next release.

adamsp, 2016-09-05:
Perfect, that'll do nicely! Thanks a lot :)

adamsp, 2016-09-07:
This update has also removed the latitude/longitude entries from the `properties` object in each event in the response. I've switched to using the `geometry.coordinates` values.

I have a couple of other questions:
- that time format seems to be in UTC, however without the timezone information on the timestamp it's difficult to tell (I've been operating under the assumption it is, and things seem to be working)
- you say this issue will be fixed in the next release; what changes would that entail? I'd prefer to avoid pushing another update if I can!
- next time there are breaking API changes, where could I get warning about this so I can be ready?

Thanks a lot for your help!

nbalfour, 2016-09-08:
Hi Adam, 
Sorry, I can't answer all your questions. Perhaps @bpeng can help with the 2nd one. 
We did try to notify our users about this change but unfortunately we don't have contact details for all our users. If you could sign up to this email list http://lists.geonet.org.nz/mailman/listinfo/updates it will help us keep you informed. 
Yes, the time zone is UTC. 
Thanks

bpeng, 2016-09-08:
Hi Adam, 
There won't be major changes to the output, probably just the time format you have mentioned. BTW do you want the ```sortby``` parameter and the ```latitude, longitude``` as properties?
Thanks

adamsp, 2016-09-09:
If you bring them back I won't complain! `sortby` is nice - it means I can build the "paging" functionality client side and avoid overloading clients with massive chunks of data. I've already changed to use the lat/long inside the `geometry` object instead, so I'm not too bothered by that.

I'll make sure to sign up to the mailing list to be informed of future changes.

Thank you all for your prompt responses and help with this! :)

bpeng, 2016-09-12:
@adamsp Update is now available with the time format in your query, and the ***sortBy*** parameter is added in the ***cql***, please check:
```
http://wfs.geonet.org.nz/geonet/ows?service=WFS&version=1.0.0&request=GetFeature&typeName=geonet:quake_search_v1&outputFormat=json&maxFeatures=50&cql_filter=modificationtime>2016-03-27T19:34:23.565Z+AND+eventtype=earthquake+AND+sortBy=modificationtime
```

adamsp, 2016-09-13:
This is great! Thanks a lot @bpeng.

closed 2016-09-13 by adamsp