GeoNet/help #16: Instrument response problems with strong motion station GKBS

shicks1988, 2016-09-15:
Hello

I'm looking at strong motion waveforms from the M7.1 offshore Gisbourne earthquake on 01/01/16.

I am correcting for instrument response using Obspy with the datelessSEED (RESP) files from ftp://ftp.geonet.org.nz/seed/RESPONSE/ - 

[RESP.NZ.GKBS.20.BNZ.txt](https://github.com/GeoNet/help/files/474006/RESP.NZ.GKBS.20.BNZ.txt)
[RESP.NZ.GKBS.20.BN2.txt](https://github.com/GeoNet/help/files/474008/RESP.NZ.GKBS.20.BN2.txt)
[RESP.NZ.GKBS.20.BN1.txt](https://github.com/GeoNet/help/files/474007/RESP.NZ.GKBS.20.BN1.txt)

The raw waveforms clearly show the earthquake, but when I correct for instrument response, the waveforms completely lose character and show a lot of spurious high frequency bursts - see attached images - 

Uncorrected
[image]

Corrected
[image]

I have tried the Obspy instrument response method using other GeoNet stations and with the provided RESP files, so I don't think there is a problem with the algorithm.

Therefore, I think it is possible that the poles and zeros in the RESP file may be incorrect. Could you please look at these files to check if they are valid for the strong motion instrument.

ozym, 2016-09-15:
Hi,

There's always a chance that we have a mistake in there somewhere so it's good to check. The instrument is a Kinemetrics Basalt running an internal EpiSensor with a peak-to-peak setting of 2g (matching +/- 5 volts).

Page 295 of the manual gives the EpiSensor response as: (I've had to reformat the equation).
> This model can be represented as V(s)/A(s) = k1 * k2 / (s - p1)(s - p2)(s - p3)(s - p4)
> where k1 = 2.46 x 10^13
> k2 = Sensitivity of sensor in V/g (1.25V/g)
> s is the Laplace transform variable
> p1 = -981 + 1009i (Pole 1)
> p2 = -981 - 1009i (Pole 2)
> p3 = -3290 + 1263i (Pole 3)
> p4 = -3290 - 1263i (Pole 4)
> V (s) is the Laplace transform of the output voltage
> A (s) is the Laplace transform of the input acceleration

The IRIS nominal response library for this instrument sampled at 40 Hz (will have same Poles/Zeros but slightly different FIR filters) can be found at:

http://ds.iris.edu/NRL/sensors/kinemetrics/RESP.XX.NS213..BNZ.Episensor.DC_200.5VDF.2G

This response is very flat to acceleration below 50Hz so it shouldn't have much of an impact. The one thing that can happen is that long period noise (from the instrument/sensor) can be mapped into the output response. The usual technique for dealing with this is to apply a bandpass filter (outside the frequencies of interest).

The other thing that might be giving problems is the sampling rates, we record two rates 50 Hz continuous (denoted by BNZ, BN1, BN2 as per your upper plot), and 200 Hz triggered (denoted as HNZ, HN1, HN2 as per your lower plot).  The response for the 200Hz will have different FIR filter settings (but the same poles/zeros),

For reference here's the automatic output assuming a flat response to acceleration over the passband and assuming the anti-alias filters are outside this passband. It will be using the 200 Hz data.

[20160901_163830_GKBS_20.PDF](https://github.com/GeoNet/help/files/474211/20160901_163830_GKBS_20.PDF)

-- Mark

nbalfour, 2016-09-15:
Hi,
I have looked at this site for the East Cape event and hadn't noticed a problem with the instrument response. I didn't use the RESP files though but instead I used the "attach_response = True" option when requesting the waveforms. Here is the code I used and the output.

`#!/usr/bin/env python

from obspy import UTCDateTime
from obspy.clients.fdsn import Client as FDSN_Client

client = FDSN_Client("GEONET")
t = UTCDateTime("2016-09-01T16:37:00.000")
st = client.get_waveforms("NZ", "GKBS","20", "?N?", t, t + 300,attach_response=True)
pre_filt = (0.005, 0.006, 30.0, 35.0)
st.remove_response(output='ACC', pre_filt=pre_filt)
st.plot()`

Corrected (output of python code above)
[image]

@ozym Could there be a difference between the FDSN response and the RESP files on our ftp?

ozym, 2016-09-15:
@nbalfour hopefully not, they're both obtained from the same dataless seed volume, although they do have different synching mechanisms.

ozym, 2016-10-10:
Close this for now - reopen if pain persists.

closed 2016-10-10 by ozym