GeoNet/help #18: Strong motion evt and text files

carolinelittle, 2016-10-05:
I'm helping a user with old raw, triggered strong motion data from 2005.

There are two files for each event, an evt and txt file. Are these the same data but different formats? There's not much in the text file. 

If they're different, can someone point me to a way he can convert the .evt file to text?

This is the quake he's looking at:

ftp://ftp.geonet.org.nz/strong/raw/altus/2005/2005.183/2005.183.0446.15.TACS.03366.evt 
ftp://ftp.geonet.org.nz/strong/raw/altus/2005/2005.183/2005.183.0446.18.TACS.03366.txt

ozym, 2016-10-05:
The TXT file was used when comms were very slow. Basically it's a summary of the event and was sent first, the full data in the EVT file was then sent.

It was meant to be used for alerting etc.

ozym, 2016-10-05:
Depends, could try the V1A formatted files or there are tools from Kinemetrics.  We also have MiniSEED but I'm not sure that would help.

carolinelittle, 2016-10-05:
It's too small a quake to have made any data in the /Proc files (I think!). 

Thanks! This might help him then: 
http://www.kinemetrics.com/p-151-SMA.aspx

ozym, 2016-10-05:
or http://wiki.kmi.com/wiki/images/5/58/K2ASCII.zip

ozym, 2016-10-05:
May need sign in from KMI to see it though.

carolinelittle, 2016-10-05:
Thanks all, @nbalfour has also done a write up of possible tools. I'll send him the options.

closed 2016-10-05 by carolinelittle