GeoNet/help #20: GeoNet website "Felt Report" page does not activate "Location Suggestion List" in Chrome

Falconne, 2016-11-13:
I have my adblocker disabled on the GeoNet site, so it's not a script blocking issue.

Typing my address in manually did not allow submission, as the site popped up a dialog asking me to select a location from the Suggestion List, which will not appear.

quiffman, 2016-11-13:
Thanks @Falconne, we're looking in to it.

In the mean time, we can still accept reports via the mobile apps.

Leftcoaster74, 2016-11-13:
Not working for me either even after my address suggestion popped up it wouldn't accept it. 217 High Street, Greymouth. Felt 12:02 earthquake and two aftershocks so far

quiffman, 2016-11-13:
Sorry folks, the Google geocode API ran out of juice - we're rolling out updated keys now.

Hopefully it'll start coming right, soon.

gclitheroe, 2016-11-13:
Thanks all.  It's sorted now and we're keeping an eye on it.

closed 2016-11-13 by gclitheroe