GeoNet/help #21: Location still not accepted by felt.geonet.org.nz

DevCon99, 2016-11-14:
Earlier, I did not even get location suggestions for the location field at felt.geonet.org.nz.

Now, I receive location suggestions, but when I select the correct one, the form still requests that I select a location from the list.

quietone1, 2016-11-14:
Same here. I've entered a couple of reports since last night. And now can't do any.  I get a popup "please select a location form the suggestion list" but there are no suggestions.

Tried with Firefor ESR 45.4.0 and Chromium 53.0.2785.143

stadtwald, 2016-11-14:
It looks to me like the Google Maps API, which is in use on felt.geonet.net.nz, is rejecting requests due to an API request limit being reached.  While debugging, I encountered the error message "You have exceeded your daily request quota for this API.  For more information on usage limits and the Google Maps Javascript API services please see: https://developers.google.com/maps/documentation/javascript/usage".

stadtwald, 2016-11-14:
I should note that the relevant API endpoint is "https://maps.googleapis.com/maps/api/place/js/AutocompletionService.GetPredictions?1s22&4sen-US&7scountry%3Anz&9sgeocode&15e3&key=AIzaSyAfr6A5wp4EAvCM10GAi3QCkdPorQezl0M&callback=_xdc_._40a1gz&token=6909"

quiffman, 2016-11-14:
Thanks all, @stadtwald is correct. We're hitting a quota limit on the API. We're switching API keys, _again_.

gclitheroe, 2016-11-14:
thanks for the report.  @stadtwald is correct.  We keep hitting the Places API request limit.  I've got a support request in with Google  to increase the limit.  In the mean time we rolling between API keys to nurse it along.  Should be ok now, but only for a while :-(

Please do try the Android or iOS apps, these don't have this problem.  http://info.geonet.org.nz/display/home/Information+at+your+fingertips

gclitheroe, 2016-12-06:
As an interim update - we ended up writing our own geocoding api using open data from LINZ.  It looks to be preforming very well.  Hopefully the issues mentioned here have gone away.

We need to do some tidy up on the build for the geocoder and the we will open up the repo.  It may take until next year to do the tidy up part.

salichon, 2019-08-22:
@carolinelittle Hello! 
 Perhaps a bit of  ticket house cleaning
I was cleaning some of mine ...  
cheers
jerome

closed 2019-08-23 by carolinelittle