GeoNet/help #23: EQ Felt filter goes off region intensity but intensity is coloured

Bwooce, 2016-11-18:
http://www.geonet.org.nz/quakes/felt/strong For example shows EQs with a regional intensity of strong+ not an intensity of strong+

It might be the right filter condition, you're the experts, but the horizontal filter bar colours are the same as the tags on the individual earthquakes for the plain intensity. 

Hard to explain in text, I have some screenshots(mobile) if you want. 

Kia kaha,
Bruce

gclitheroe, 2016-11-18:
thanks for the issue.  Unfortunately it is terribly confusing!  We have been working to update the web site.  This weeks quake delayed this.  I hope we will get back to the update work before the end of the year.

In the mean time if you can use the mobile apps the filtering is clearer in the apps

http://info.geonet.org.nz/display/home/Information+at+your+fingertips

salichon, 2021-11-22:
consider this resolved since the time ...

closed 2021-11-22 by salichon