GeoNet/help #24: Felt Strong & severe miss latest quake

stephenzan, 2016-11-18:
Using the Felt feature of the GeoNet Quakes website, when I choose Strong or Severe the latest entry is usually missing. It appears in the next lower strength. i.e. when choosing Severe, the latest severe quake is missing. When choosing Strong, the latest strong quake is missing.

Pages are:
https://www.geonet.org.nz/quakes/felt/strong
https://www.geonet.org.nz/quakes/felt/severe

I am using Firefox on Windows 10 PC.

go well, Stephen

gclitheroe, 2016-11-18:
It is very confusing and we are working to fix this.  Please see #23 as well.

stephenzan, 2016-11-19:
Yes, it is the Regional Intensity being filtered which is confusing. Thanks for that. #23 is a better description.

closed 2016-11-19 by stephenzan