GeoNet/help #25: GeoNet Android App Alarm not sufficient to wake user up at night!

teverth, 2016-11-19:
The GeoNet App is great. But the offered audible notifications are insufficient to wake a person up at night. Can you allow for the setting of an alarm clock type notification that keeps ringing until turned off or for a minimum of say 1 minute?
We live in Whitianga and having a fast and reliable wake up call in case of a severe earthquake could be important due to the tsunami risk for our community. 
BTW: The latest Kaikoura quake's Tsunami alert response was pathetic. The local sirens were triggered a full 2.5 hrs after the 7.8 quake, much too late to get people to safety in case there had been a tsunami risk. But also, the alert proved to be unnecessary as there was no tsunami and probably no risk. (Low tide, far away). The fact the sirens went on caused distress and several car crashed in town in the ensuing rush to high ground.

nbalfour, 2016-11-21:
Thank you for your feedback. Our app is not designed to be used as a warning for earthquake and tsunami. It's purpose is to notify you on earthquakes that have occurred in New Zealand at a level (either magnitude or intensity) based on your notification settings. Depending on the brand of phone you have will determine the sound options available. 
Local Civil Defence groups are responsible for Tsunami Warnings in your local area.

teverth, 2016-11-22:
Thank you,
    I would still think that the possibility of
        being woken up by your app would ad a significant layer of safety
        for those like use in Whitianga, who live in a clear Tsunami
        hazard area.
    The time between an event that could create
        a tsunami and the reaction of our local warning authorities is
        currently unpredictable. At the Kaikoura quake the local alarms
        rang a full 2.5 hours after the quake! If there had been a real
        danger for us, this would have been too little too late. 
      
    The time it takes to alert an off duty
        GeoNet scientists at night to assess a new quake to alerting
        also off duty Civil Defence groups and to then come to a decision
        to activate the alarm system will lose the very minutes we have
        to get to safety in case of a quake at the Hikurangi Trough or
        further North. Also, in many rural coastal communities Tsunami
        sirens are not audible at all.
    The best hope really is a direct alert
        notification such as the ones send out by your app. But as I
        said, the brief 'bing' the user can select is not sufficient to
        wake a sleeping family up. It would be a small request for the
        developer I guess to add the option to set the notification to an
        alarm clock ring that continues until cancelled.
    
        Best
    
        Thomas Everth
      
    
    
    
TE-Software Ltd
Thomas Everth
103 Robinson Road
Whitianga 3510 / New Zealand
Tel.: (+64)-7-866-5507
Mob.: (+64)-275-947133
Skype: teverth

    On 22/11/2016 11:37 AM, Natalie Balfour
      wrote:
    
    
      Thank you for your feedback. Our app is not designed to be used
        as a warning for earthquake and tsunami. It's purpose is to
        notify you on earthquakes that have occurred in New Zealand at a
        level (either magnitude or intensity) based on your notification
        settings. Depending on the brand of phone you have will
        determine the sound options available.
        Local Civil Defence groups are responsible for Tsunami Warnings
        in your local area.
      —
          it on GitHub, or mute
          the thread.

closed 2021-10-05 by salichon