GeoNet/help #26: Quake filter does not filter current datasets

fb3rasp, 2016-11-22:
Hi,
I love geonet.org.nz and I'm visiting the site very often recently. 
I noticed that the filter bar (weak+, light+, moderate+, strong+, severe) under 'felt quakes' seem not to filter on the most current dataset.
I see (when selecting 'weak' a moderate quake which dates occured about 10mins ago. When I select 'moderate+' this quake does not appear. 
Same applies for the other filters.
'moderate+' shows quakes which occured 7 hours ago, no quake in the last 7 hours are listen, but appear on the 'weak+' setting.

Cheers

gclitheroe, 2016-11-22:
Thanks for the issue.  It is very confusing.  It's something we were working on before the quake.  More info in this issue #23

salichon, 2021-11-22:
Consider this resolved since the time ...

closed 2021-11-22 by salichon