GeoNet/help #28: Cannot submit a felt report.

radar57, 2016-12-04:
Hi there. I am currently situated in Waikanae. In New Zealand. But for some reason the GeoNet iPad app seems to think I am in the US.  So I get this weird error every time I try to submit a Felt Report. It has only started doing this recently. The error reads as follows:

Domain=com.amazon was.AWSCognitoIdentity ErrorDomain="(null)" UserInfo={message=Access to Identity 'us-east-1:5de046a7-ffb6-4306-9f41-351d 198cf941' is forbidden., _type=NotAuthorizedException}

junghao, 2016-12-04:
Would you please try remove GeoNet app and reinstall it? I'm guessing your app is stuck on previous login session.

radar57, 2016-12-07:
Thank you for your response. I have uninstalled and reinstalled several times already but still get the same error. I have only ever submitted a felt report using my iPad from my home address in Waikanae so cannot think how it would confuse that with US.  I switched Location Services off and it asked me to turn it on so if it is using that and not a previous login, how can it not see where I am? 

Thanks for your help

junghao, 2016-12-15:
Hi @radar57 ,
The error has nothing to do with your location so please keep your location service as usual.
We've released a new version 2 days ago. Would you please install it and see if it resolves your issue?

closed 2021-10-05 by salichon