GeoNet/help #29: FDSN down 07/12/2016

calum-chamberlain, 2016-12-06:
Hello all,

I'm unable to download from the GeoNet FDSN service (via obpsy) this morning and overnight.  I guess and hope that this is temporary?

calum-chamberlain, 2016-12-06:
Looks like it's up again - out of curiosity, is this something that we can expect often, and is there an obvious cause? I have seen it go down quite often (but not reported it because it comes back), but it ends up failing tests on EQcorrscan that rely on downloading data.  It also means that I can't rely on running aftershock detections etc downloading data on-the-fly...

nbalfour, 2016-12-06:
Hi Calum, 

Yes, FDSN went down last night and has been restarted. 

We are currently in a bit of a transition and GeoNet is in the process of appointing a new Data Management Team Leader. This person will be responsible for reviewing our data services and the associated support levels. 

We have different support levels (aka tiers) for our services and they are monitored/repaired according. 

Unfortunately, FDSN currently doesn't have a support level defined but as I mentioned this will change with the incoming Data Manager. 

Please keep in mind that anything built on GeoNet data services will only be as reliable as the support level/tier for that service.

So, in conclusion, yes, we understand this is a problem and it will be looked at in more detail in near future. 

Thank you for bring the problem to our attention.

closed 2016-12-07 by nbalfour