GeoNet/help #30: Finding the peak acceleration from an earthquake sequence

timbvt, 2016-12-11:
Good morning, 

I'm trying to get some data from the strong motion data records and want to make sure that I have found the largest in the timeframe without trawling through every record.

Is there a way to find the largest strong motion data set for a set time period? Something similar to a summary plot showing the peak acceleration of many different data sets so that you can locate the peak acceleration for a given earthquake sequence? Specifically, I am looking for the peak acceleration from the Kaikoura earthquake that occurred at the Culverden Airlie farm site.

Any help would be much appreciated.

Cheers

carolinelittle, 2016-12-20:
Sorry for the slow reply @nbalfour can you please reply to this?

nbalfour, 2016-12-21:
I believe this was resolved in an email thread that I replied too. I will close the issue and it can be reopened if there are more questions.

closed 2016-12-21 by nbalfour