GeoNet/help #31: Address missing from drop-down list

MarkDotBlair, 2017-01-01:
I wanted to submit a Felt report, but your list does not show any version of my address, which is 4/36 Queen Street, Mount Victoria, Wellington.

MarkDotBlair, 2017-01-01:
This is on your website https://felt.geonet.org.nz/

carolinelittle, 2017-01-09:
Hi Mark, you're right, it doesn't recognise your address for some reason. It's fine with 34 Queen Street, Mount Vic - and we aggregate the addresses, so I'd suggest using your neighbour's address. 

Sorry about this. We are working on a building our own search with a LINZ database, so hopefully we can fix these type of issues in the future (we currently use Google).

@sorennh or @bpeng can you shed any light?

MarkDotBlair, 2017-01-09:
OK, thanks, it's not just me then :-)

Best wishes sorting it.

Cheers,
Mark

Tumoana, 2017-04-09:
Yea me too. 2/139 Wadestown road, Wadestown, Wellington 6012. Google aint googling enough :)

carolinelittle, 2017-04-09:
Hi @Tumoana, yes, you're in the same boat. Google doesn't know that 139 is more than one house. It'll accept 139 Wadestown Road, which will work fine with our aggregation of locations.

Niteowlnz, 2017-05-06:
Hi   same for me 10 Goodman Drive , Motueka

sara-page, 2017-05-30:
Hi @Niteowlnz  sorry for the delay.  if you use 11 Goodman Drive, it will work.

closed 2021-11-22 by salichon