GeoNet/help #32: FDSN 'event' data?

filefolder, 2017-01-11:
Hi all

I know the FDSN portal states that only "station" and "dataselect" are available to query but I also see that the "event" portal is UP though I can't seem to retrieve anything from it. Is this feature forthcoming? Hopefully?

As a seiscomp3 user my issue is that IRIS does not supply events in sc3ml format and converting from QML is a becoming a larger and larger headache, not to mention there seems to be some inconsistencies between the IRIS data and what is available direct from GeoNet.

If there's anyway to get access to more GeoNet stuff in sc3 format, particularly event data, that would be very helpful. Is there a method other than FDSN I should look at?

Thanks!

gclitheroe, 2017-01-11:
Hi,

> I know the FDSN portal states that only "station" and "dataselect" are available to query but I also see that the "event" portal is UP though I can't seem to retrieve anything from it. Is this feature forthcoming? Hopefully?

It is on the way (I was writing code for it today).  I think it will be the order of weeks to months away.  However, we don't get to plan all our work priorities (earthquakes etc).

In the mean time you can get at QuakeML here http://quakeml.geonet.org.nz/quakeml/1.2/2012p564088
and the corresponding SC3ML here (note the `.xml` on the end compared to getting QuakeML) http://seiscompml07.s3-website-ap-southeast-2.amazonaws.com/2012p564088.xml

The QuakeML is generated on the fly from the SC3ML with the XSLT from http://geofon.gfz-potsdam.de/ns/seiscomp3-schema/0.7/sc3ml_0.7__quakeml_1.2.xsl

Please keep in mind that the http://seiscompml07.s3-website-ap-southeast-2.amazonaws.com... URL changes when we do an SC3 upgrade that also has a schema change.  There is one of these pending, probably March - April.  When we implement the FDSN event service there will be a more stable URL to get SC3ML from.

If you want a lot of event SC3ML and you're comfortable with the AWS CLI then let me know and you can sync from the S3 bucket (or you can guess it from the above URL).

Have you found either WFS or quakesearch for searching for events?

* https://wfs.geonet.org.nz/
* https://quakesearch.geonet.org.nz/

Hope this helps,
Geoff

filefolder, 2017-01-12:
Thanks, that's great to hear. I can probably wait but in the meantime I may try to pull a hundred or so events from that link, which I don't imagine should be a huge strain.

n.b. the newest seiscomp release (2016.333) now defaults to the 0.9 sc3ml schema; there doesn't seem to be an easy way to convert 0.8 or 0.9 data but directly importing 0.7 into it seems fine at the moment.

gclitheroe, 2017-01-12:
Pulling hundred will be absolutely fine.

If you do need to convert 0.7 to 0.8 you can just edit the SC3ML and change the versions at the top:

```
<seiscomp xmlns="http://geofon.gfz-potsdam.de/ns/seiscomp3-schema/0.7" version="0.7">
```

to 

```
<seiscomp xmlns="http://geofon.gfz-potsdam.de/ns/seiscomp3-schema/0.8" version="0.8">
```

which then validates against the 0.8 xsd (you will need the `quakeml_types.xsd` in the same directory.

```
xmllint --noout --schema sc3ml_0.8.xsd 2015p768477_0.8.xml 
2015p768477_0.8.xml validates
```

I'm using files from http://geofon.gfz-potsdam.de/ns/seiscomp3-schema/

And I can see that going from 0.7 to 0.8 is all optional additions:

```
diff sc3ml_0.7.xsd sc3ml_0.8.xsd
3c3
< <xs:schema xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:qml="http://quakeml.org/xmlns/quakeml/1.0" xmlns:scs="http://geofon.gfz-potsdam.de/ns/seiscomp3-schema/0.7" targetNamespace="http://geofon.gfz-potsdam.de/ns/seiscomp3-schema/0.7" elementFormDefault="qualified" attributeFormDefault="unqualified">
---
> <xs:schema xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:qml="http://quakeml.org/xmlns/quakeml/1.0" xmlns:scs="http://geofon.gfz-potsdam.de/ns/seiscomp3-schema/0.8" targetNamespace="http://geofon.gfz-potsdam.de/ns/seiscomp3-schema/0.8" elementFormDefault="qualified" attributeFormDefault="unqualified">
772a773,783
>   <xs:complexType name="ResponseFAP">
>     <xs:sequence>
>       <xs:element name="gain" type="xs:double" minOccurs="0" maxOccurs="1"/>
>       <xs:element name="gainFrequency" type="xs:double" minOccurs="0" maxOccurs="1"/>
>       <xs:element name="numberOfTuples" type="xs:integer" minOccurs="0" maxOccurs="1"/>
>       <xs:element name="tuples" type="scs:RealArray" minOccurs="0" maxOccurs="1"/>
>       <xs:element name="remark" type="scs:Blob" minOccurs="0" maxOccurs="1"/>
>     </xs:sequence>
>     <xs:attribute name="publicID" type="qml:ResourceIdentifier" use="required"/>
>     <xs:attribute name="name" type="xs:string"/>
>   </xs:complexType>
926a938
>       <xs:element name="responseFAP" type="scs:ResponseFAP" minOccurs="0" maxOccurs="unbounded"/>
```

I don't actually see the 0.9 files being available yet.  We will chase the about this.

nbalfour, 2017-08-06:
@filefolder the event service is now in beta. If you would like to use it please use the client https://beta-service.geonet.org.nz 
You can read more about it on our website http://www.geonet.org.nz/data/tools/FDSN 
If you have any issues, please raise them in this repo [GeoNet/fdsn](https://github.com/GeoNet/fdsn/issues)

nbalfour, 2017-11-08:
@filefolder the event service is now in production at service.geonet.org.nz please see http://www.geonet.org.nz/data/tools/FDSN for the most up to date information on our FDSN webservices.

closed 2017-11-08 by nbalfour