GeoNet/help #34: no station or response data available for OTAZ

filefolder, 2017-01-30:
hi there--

can't seem to FDSN any metadata for OTAZ, which seems to be an old auckland area station is no longer up. waveforms are available and a lot of events in the mid 00's reference OTAZ but can't find any further info. not a huge problem but hopefully it just got misplaced somewhere?

thanks!

calum-chamberlain, 2018-12-19:
I wanted to point out that this seems to be an issue for other sites and is becoming a bit of a pain. For example sites CNZ and MGZ appear to have been active until 2005, but do not appear on the station search online, nor do they return response or channel information from an FDSN query. e.g.:

```python
from obspy.clients.fdsn import Client

client = Client("GEONET")
client.get_stations(station="MGZ")
"""
Inventory created at 2018-12-18T02:39:42.000000Z
	Created by: Delta
		    
	Sending institution: GeoNet (WEL(GNS_Test))
	Contains:
		Networks (1):
			NZ
		Stations (1):
			NZ.MGZ (Maungaku)
		Channels (0):

"""
client.get_stations(station="MGZ", level="response")
"""
FDSNNoDataException: No data available for request.
Detailed response of server:

"""
```
The same happens for `level=channel`. 

It looks like the meta-data available through the FDSN doesn't match the waveforms that are available prior to about 2008/2009 very well. 

Other examples include:
- WEL accelerometer (location code 20) only had response information from 2008-12-18, but I have downloaded waveform data from at least 2005 for that instrument
- TAZ has metadata for channel EHZ for a month in 2007, but waveform data are available for TAZ SHZ in 2005
- No response info for 2005 for NZ.UTU.10.SHZ... The list is quite long.

By the way, the reason I'm running into these issues is because I have downloaded the geonet catalog since 1987, then downloaded waveform data for every pick that was available and I'm now trying to remove the responses for them so that waveform data can be compared from the same site across different epochs.

ozym, 2018-12-19:
@calum-chamberlain with a few exceptions, you are bumping into the analogue telemetered sites where the meta data still needs some work. These are mostly in the volcanic regions (and Wellington) and are usually identified by 50 Hz and mostly vertical (SHZ). They could also be recorded at multiple places (due to the telemetry used) so there may be different versions of the same data (10, 11, ...).

The original metadata was based on triggers used in the processing (start / stop times), however we have now attempted to load in all data recorded which may include triggers not used or times when the sites were recording data but outside the estimated range.

Note, due to the nature of the telemetry in most cases we actually won't know what the full responses are. It may be possible to invert back for gain ....  but that's another project.

This has been quite a process, we're slowly working back in time. You can see our full metadata archives in https://github.com/GeoNet/delta the stationxml that is returned from FDSN is built from this repo.

calum-chamberlain, 2018-12-19:
Oh man that sounds painful to try and do.

It looks like the national network sites have meta-data as far back as they can at the moment, does that sound right?

Thanks again for the response @ozym!

ozym, 2018-12-19:
Yes, the National network sites have been easier as they had been single boxes and single sensors prior to the broadband upgrades. There have been exceptions however, the sensors deployed and recorders used at WEL have been very complicated over the years (even including telemetry) ....

calum-chamberlain, 2018-12-19:
Thanks for that Mark, that is really useful to know. I will use what I have which still seems to be a large chunk of the data.
Thanks!

salichon, 2019-08-18:
@ozym 
I assigned Muriel and myself as a reminder of this for the data team :).
Cheers
jerome

salichon, 2019-12-18:
OTAZ is a station closed in 2009 and has  

- 10.EHZ  data from 2001 to 2009
- 10 SHZ data from 1996 to 2001 
- Sensor is a Sercel Inc.,L4C over this Period of time 
- Metadata to be completed as a legacy into the Delta repository

so far fdsn returns this 
[https://service.geonet.org.nz/fdsnws/station/1/query?station=OTAZ&level=station](https://service.geonet.org.nz/fdsnws/station/1/query?station=OTAZ&level=station)

-Other metadata Legacy to be addressed or documented  back in the past (SH) streams  like UTU, TAZ  (see comments of Mark above about the the telemetry) 
 
FYI @JonoHanson

salichon, 2021-11-22:
Will need investigation to complete metadta in the past

closed 2021-11-22 by salichon