GeoNet/help #35: FDSN down

calum-chamberlain, 2017-03-01:
Hi all,

Looks like the FDSN is down this morning, just wanted to check that you knew.

Accessing via ObsPy's FDSN Client, error output as follows:

```python
  File "/home/calumch/my_programs/Python_virtual_envs/eq_2/local/lib/python2.7/site-packages/eqcorrscan/tutorials/get_geonet_events.py", line 96, in _get_geonet_pubids
    client = Client('GEONET')
  File "/home/calumch/my_programs/Python_virtual_envs/eq_2/local/lib/python2.7/site-packages/obspy/clients/fdsn/client.py", line 258, in __init__
    self._discover_services()
  File "/home/calumch/my_programs/Python_virtual_envs/eq_2/local/lib/python2.7/site-packages/obspy/clients/fdsn/client.py", line 1482, in _discover_services
    raise FDSNException(msg)
FDSNException: No FDSN services could be discovered at 'http://service.geonet.org.nz'. This could be due to a temporary service outage or an invalid FDSN service address.
```

nbalfour, 2017-03-02:
Hi Calum, 
FDSN has been restarted and should be working now. 
FYI, we are currently developing a new FDSN service that will be supported. It will included all 3 functions and will be much faster. So watch this space. 
Thanks,
Natalie

calum-chamberlain, 2017-03-02:
Yay, thanks! Would be happy to help/test things (my EQcorrscan tests ping the GeoNet FDSN everytime they run).

nbalfour, 2017-03-02:
@crleblanc, please note Calum's use for FDSN might help for testing the new service ^^

closed 2017-03-02 by nbalfour