GeoNet/help #36: Unclear information about polarity reversals of stations

caro-github, 2017-03-02:
Hi Geonet team,

as discussed with Natalie, I am submitting this here so that other people can profit from the response as well.

On the ftp site (under POLARITY) information on polarity reversals can be found, such as the following:
NZ RPZ   20 HNZ 2003-01-03T00:00:01 2015-10-14T23:00:00   0.0  90.0 -1.0

However, it is not clear whether that relates to the raw data from the station or wrongful response files.
Looking at teleseismic events for RPZ it is clear that the instrument response function causes the reversed polarity during the above listed period, while the raw data of the actual station components are consistent with e.g. FOZ and WVZ and not reversed.

The station that I am struggeling with, however, is ODZ. The following information is provided:
NZ ODZ   10 HHZ 2003-09-11T06:00:00 2006-10-13T04:20:00   0.0  90.0 -1.0
NZ ODZ   10 HHZ 2006-10-13T04:20:01 2008-07-08T00:00:00   0.0  90.0 -1.0

From looking at the rawdata of the following telseismic events I have identified: 
2006/09/09 4:21 M6.3 ODZ vertical component correct polarisation (which contradicts the above)
2006/11/14 14:29:29 M6.3 ODZ vertical component reversed 

So I am assuming that a change took place on 2006-10-13. Could you please specify what change this was, and whether my observations are correct or provide a counterexample that supports  the Geonet information.

Thanks
Carolin

ozym, 2017-03-02:
Hi Carolin

In general the polarity information, as given in the POLARITY files, indicates that there is a systematic reversal between how this instrument is installed and what is recorded in the archive. 

The response files are created so that the raw ground measurements can be recovered from the digital counts stored in the archive.  In some cases the sites have been wired so that the signals are reversed, it is easier to maintain this reversal information rather than alter the dip and azimuth of the sensor.

In some cases it is the sensor itself (i.e. some sensors give a +ive signal for a "downward" ground motion) but in most cases it represents a cabling issue.  We have some cables which are consistently reversed when connected to particular dataloggers (due to their construction). The most obvious case is the STS-2 sensor connected to a Q4120 datalogger.  The azimuth and dip information for the installed sensor will be given as measured, the reversal, if present, is then applied to rotate the sensor 180 degrees (virtually) as shown in the polarity files. 

For your specific comments above:

At RPZ the strong motion sensor (HNZ) was  replaced with a Basalt recorder on 2015. Prior to that it was standalone sensor connected to a Q4120 with a consistently reversed cable.

ODZ is trickier, the sensor is an STS-2 and was replaced with a similar model in 2006-10-13. The datalogger itself was not replaced until 2008. Any reversal would indicate a problem with that particular replacement cable and would need input from the technical team as to what was changed.

Hopefully this helps, polarity measurements and checks have been difficult especially after the equipment has been removed! 

Mark

nbalfour, 2017-03-02:
I can't answer all of your questions right now but from my initial assessment of ODZ it appears that it was the sensor itself changed on 2016-10-13.

Perhaps @ozym, @jristau might be able to provide more information.

caro-github, 2017-03-02:
Thank you very much for this information. To me the sensor change at ODZ is associated with a reversal of the polarity from correct to reversed. Serendipitously, the teleseismic event on 2006/11/13 1:38 confirms this as well.

In my previous comment I was actually referring to the seismometer at RPZ (HHZ), but copied the wrong line (as you may have guessed, correct one below). Anyway, I have many examples to show that RPZ HHZ is consistent with FOZ and WVZ HHZ and I think the "reversal" is only due to the response file. 

NZ RPZ   10 HHZ 2003-01-03T00:00:01 2015-10-14T23:00:00 185.0  90.0 -1.0

salichon, 2021-11-22:
- this will be addressed into the Delta repository https://github.com/GeoNet/tickets/issues/9289

closed 2021-11-22 by salichon