GeoNet/help #37: Annette Plateau (NETT) cGPS

elidana, 2017-04-03:
From Paul (Otago University):

> NETT is in a (reasonably) good location in respect of the SAGENZ profile. It is close to the main divide and sufficiently high in elevation to avoid significant masking.  Given it’s elevation, it has actually operated reasonably well since it gets good sun and wind to keep the pv panels clear and the batteries topped up.
>
> But the large horizontal seasonal biases is a nuisance.  Although it can be modelled, I am concerned about the close proximity of the site to a large drop towards the east (Tasman Valley side).  Given the level of horizontal movement, it is probably a matter of time before a block of rock falls towards the Tasman Valley.  Obviously I am in no position to predict when this could happen.
>
> I thought it was worth considering if an alternative site could be found.

And a further comment from Sigrun (GNS Science):

> Absolute agree with Paul that it would be good to relocate.  I am not so concerned about the horizontal as I think that is just an affect of where the site is (see it at a lot of similar sites).
> I am more concerned with the vertical we see at the site.  it is not moving up at the rate I would expect which to me suggest the site might be unstable.
> Also.  The site has been in operation now for many years.  It would be nice to get a well constrained velocity at another location that might help with modeling and hopefully avoid the seasonal and low vertical observed at NETT

elidana, 2017-04-03:
similar to [GUNR problem](https://github.com/GeoNet/help/issues/9)

while we discuss the relocation/rebuild might be worth clarifying with landslide people if there's any interest in keep recording data from these two sites

cpearson86, 2017-04-04:
If Sigrun is concerned about an anomalous vertical velocity than confirming or otherwise is a good scientific reason to have another measurement point in the area. My only concern is finding another suitable site. I was responsible (with Paul in siting the original SAGE net points and good sites are not all that common. Rock is everywhere but stable rock is not. Add to that the requirements for snow free winter operations and finding a suitable site is challenging.

elidana, 2017-04-06:
@cpearson86 thanks a lot for your comment, much appreciated. 
Relocation is surely challenging in the area, and won't happen anytime soon. 
At the moment, we are trying to collect information about sites that are showing an unexpected behavior. Any input from the scientific community and geodesists who process these data are really useful.

elidana, 2018-08-22:
The quality of raw data and post-processing daily solutions are showing that this site is not providing good quality data. Also, its geographical location is close to HORN, which is providing better results. The site was flattened in march 2017 and would need a complete rebuild. Given the not ideal quality of associated data, the site has been dismantled. The monument has been left in place in case future scientific projects in the area will need to reoccupy the site.

site | East (mm/yr) | North (mm/yr) | Up (mm/yr) | WRMS East (mm) | WRMS North (mm) | WRMS Up (mm)
-- | -- | -- | -- | -- | -- | --
HORN | -22.35 +/- 0.07 | 33.43 +/- 0.07 | 5.10 +/- 0.27 | 4.99 | 2.93 | 9.62
NETT (*) | -19.15 +/- 0.05 | 34.54 +/- 0.05 | 3.89 +/- 0.18 | 16.26 | 5.96 | 13.45

(*) velocity and WRMS are calculated only for 2010-2017. Older data are affected by non-tectonic signals