GeoNet/help #38: Accessing tide gauge data via ObsPy from the CWB_Client not working ?

elidana, 2017-04-06:
A user has problems in accessing tide gauge data through CWB

> I had successfully and repeatedly used obspy to access the CWB_Client to retrieve tide gauge data for the Kaikaura tide gauge and others.
> 
> This currently doesn't seem to work.
> 
> Here is an excerpt of the code I'm using:
> 
> ```
>         eventTime = UTCDateTime('2016-11-13T11:02:56.00Z')
>         st = self.client.get_waveforms(network="NZ",
>                                   station=stationCode,
>                                   location='41',
>                                   channel= channel, #'?T?', '*', 'BTT'
>                                   starttime = eventTime - datetime.timedelta(hours=self.prerun),
>                                   endtime = eventTime+duration*60*60)
> 
>         if bandpass:
>             print 'TG_Class: getLiveTideGuagesObsPy: de-tiding with bandpass filter and de-mean ...'
>             st.detrend(type='demean')
>             st.filter('bandpass', freqmin=1/3600.0, freqmax=1/180.0, corners=2, zerophase=True)
> 
>         data = st.merge().traces[0].data/10000.0
> ```
> 
> the problem is that `st.traces` is returned as an empty list. I've tried the LTH channel and BTT, but no luck.
> Tide gauges I tried are KAIT, SUMT.
> The server seems to be up as I do not get any error message.
> Any idea what might be going on there ?
>

seamap3d, 2017-04-06:
Thanks for posting ...

elidana, 2017-04-06:
Hello @seamap3d , you should now be able to access the data.
Could you please have a look and advise us if you are still having problem? Thanks

seamap3d, 2017-04-07:
unfortunately now I get:

extracting tide guage data for :  2016-11-13T11:02:56.00Z KAIT
17-04-07 00:10:07 Timeout on connection - try to reconnect
17-04-07 00:10:37 Timeout on connection - try to reconnect

cheers

seamap3d, 2017-04-07:
Thanks heaps guys it seems fixed now and I can access the data again ... I'll close the issue

seamap3d, 2017-04-07:
Ha ! I don't think I actually can close this ;) - would you mind ?