GeoNet/help #44: RDLV GNSS daily position

elidana, 2018-05-11:
From @pdenys162

----
Hi Elisabetta
 
Do you know if something has happened to the site RDLV.  Since the end of 2017/start of the year, the easting coordinate component appears to have become very erractic. 
 May be the antenna is failing or some obstruction has been placed near the site?
Just curious.
 
Cheers Paul
 
[image]

elidana, 2018-05-11:
Hi Paul,

very nice you spotted that. 

Our records does not show any obvious obstructions in the antenna surroundings or equipment changes that might explain that behavior.

Our routine Gamit/Globk solutions are showing same results as your processing output. The vertical component seems also affected.

*Daily position time series*
[image]
[image]
[image]

The antenna was installed back in 2009 so I do agree that the cause might well be that it is starting to fail. 

Thanks a lot for letting us know, very much appreciated!

staylorofford, 2018-05-13:
I've processed data from RDLV from when the NetR9 was installed (2016) and made a few plots of teqc metrics. In these plots each x-tick is around 100 days, and a 100 sample running mean is overlain over the data. The break in trend shown in Paul's plots occurs around hour 68800, or around the time of S1 and S2 running mean drop in the last two plots (also the only plots showing any coincident trends!).

Note: dotted vertical lines show firmware upgrade times.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

pdenys162, 2018-05-14:
Thanks Sam. 
So the SNR plots would also suggest that either the antenna is failing or that an obstruction has been created close to the antenna that affects the GNSS signal. 
The units of the SNR is db?    What does a change of one unit imply? 

Paul

staylorofford, 2018-05-14:
The units of SNR are dBHz, so a change of ten units represents a 10x difference. A change in one unit suggests a change in how easily the receiver can hold satellite locks/tracking and how much information it can extract from a given satellite's signal.

I suspect an antenna fault is more likely in this case as there is no obvious change in the multipath like what we would expect if a physical obstruction was introduced. However it is possible that an obstruction could interfere with the GNSS signal without being detectable in the multipath timeseries. Either way, we will know around noon tomorrow what the cause is of these trends is.

elidana, 2018-05-15:
Hi @pdenys162 , 

the site has been visited today and we can confirm that the "erratic behavior" was due to an antenna fault. The antenna has been replaced with a TRM57971 on 2018-05-14T22:25:48Z.

We can also confirm that there are no physical obstructions in the antenna surroundings.

We will review the time series in a few weeks time to confirm that the issue is now fixed.

pdenys162, 2018-05-24:
Positioning data has reverted to the pre-2018 position.  A change of approximately 2cm in east and height components and 5 mm in the north component. 

I notice that the antenna metadata has not been been updated in the GNSS data archive for DOY 135-137 15-17 May 2018.

134/RDLV1340.18D.Z:30473655            TRM55971.00     NONE                    ANT # / TYPE
135/RDLV1350.18D.Z:30473655            TRM55971.00     NONE                    ANT # / TYPE**
136/RDLV1360.18D.Z:30473655            TRM55971.00     NONE                    ANT # / TYPE**
137/RDLV1370.18D.Z:30473655            TRM55971.00     NONE                    ANT # / TYPE**
138/RDLV1380.18D.Z:30895612            TRM57971.00     NONE                    ANT # / TYPE
139/RDLV1390.18D.Z:30895612            TRM57971.00     NONE                    ANT # / TYPE

elidana, 2018-06-05:
Thanks @pdenys162 ,
also GeoNet time series are showing that positions are back to what is expected for the site.

[image]
[image]
[image]

Relatively to rinex headers, those were generated before the [station metadata database](https://github.com/GeoNet/delta)  was updated, and have now been regenerated with the correct header.

Once again, thanks a lot for catching up the issue!

closed 2018-06-05 by elidana