GeoNet/help #45: Different response for "response" and "channel" levels for FDSN station service

calum-chamberlain, 2018-06-26:
Requesting station information using the obspy FDSN client results in the expected response for `level=channel`, but returns all channels for all time with `level=response`.

Consider:
```python
from obspy import UTCDateTime
from obspy.clients.fdsn import Client

client = Client("GEONET")
inv = client.get_stations(
    network="NZ", station="RPZ", location="10", channel="HHZ",
    starttime=UTCDateTime(2010, 8, 20), 
    endtime=UTCDateTime(2010, 8, 21), level='channel')
assert len(inv[0][0].channels == 1)
# Passes
inv = client.get_stations(
    network="NZ", station="RPZ", location="10", channel="HHZ",
    starttime=UTCDateTime(2010, 8, 20), 
    endtime=UTCDateTime(2010, 8, 21), level='response')
assert len(inv[0][0].channels == 1)
```
The second assert Fails because four channels are returned, which extend beyond the period requested.

nbalfour, 2018-06-26:
Thanks for drawing our attention to this. 

I have checked using the URL query, such as:
`https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=RPZ&location=10&channel=HHZ&starttime=2010-08-20T00:00:00.000&endtime=2010-08-21T00:00:00.0000&level=response`

And the query above returned all time periods for the channel, like you said it did with obspy. It should only return the response that is valid for the time period requested. 

I will ask one of the team to look into this issue and if there was a reason it was implemented this way. If not, we will correct it so that it only returns the response or responses valid for the time period requested.

calum-chamberlain, 2018-06-26:
:tada: Thanks @nbalfour. Just in case anyone stumbles across this and needs a work around, `Inventory` objects have a `.select()` method, so you can get rid of extraneous information if you need to:
```python
from obspy import UTCDateTime
from obspy.clients.fdsn import Client

inv = client.get_stations(
    network="NZ", station="RPZ", location="10", channel="HHZ",
    starttime=UTCDateTime(2010, 8, 20), 
    endtime=UTCDateTime(2010, 8, 21), level='response')
inv = inv.select(starttime=UTCDateTime(2010, 8, 20), endtime=UTCDateTime(2010, 8, 21))
```

salichon, 2018-06-26:
thanks for this ! 
 I had to have a work around using obspy (creating an array of responses) to select the right time periods 
 ... the "starttime/endtime" naming was  not helping to understand.

calum-chamberlain, 2018-06-27:
@salichon It looks like that doesn't always work (and is something I will raise with obspy) but I'm not sure why - use with care!

Also agree with confusing naming, shall discuss with obspy people.

```python

from obspy import UTCDateTime
from obspy.clients.fdsn import Client

client = Client("GEONET")
inv = client.get_stations(
    network="NZ", station="WVZ", location="10", channel="HHZ",
    starttime=UTCDateTime(2010, 8, 20), 
    endtime=UTCDateTime(2010, 8, 21), level='response')
inv = inv.select(starttime=UTCDateTime(2010, 8, 20), endtime=UTCDateTime(2010, 8, 21))
# Gets a single channel, but with the wrong time-period.
print(channel.start_date)
# outputs: "2008-02-15T16:14:00.000000Z"
print(channel.end_date)
# outputs: "2011-10-10T00:00:00.000000Z"
```

closed 2019-07-17 by francovm