GeoNet/help #48: Old data availability via FDSN

calum-chamberlain, 2018-10-11:
I'm trying to see how far back through the GeoNet catalog I can get waveforms for, and I just wanted to check that waveforms are not available prior to about the 22nd of September 1986? If earlier (digital) data are available, do you know how we could access it? 

Alongside this it might be worth noting on [this page]( https://www.geonet.org.nz/data/tools/FDSN) when data stop being available, and when the archive transitions from just triggered data to including continuous data. I think it might be worth this note because the event catalog extends back much further (which is great), so just a quick note that matching waveform data are not available would be helpful.

Thanks!

francovm, 2018-10-15:
Dear Calum,

Regarding your request, there are not earlier digital data available. Prior to 1987, we have paper records for waveform data. If is required, you are more than welcome to visit Geonet/GNS Science to take a look to our paper archive.

Thanks for your suggestion.

Regards,

Franco

calum-chamberlain, 2018-10-15:
Thanks for that confirmation Franco.

closed 2018-10-15 by calum-chamberlain