GeoNet/help #49: Are GeoNet quakeml horizontal uncertainties reported in km or m?

fiikennyg, 2018-10-16:
Are the horizontal uncertainties in the GeoNet quakeml reported in km or m. I could the see the depth uncertainties are reported in m. 

E.g. for resource_id: smi:nz.org.geonet/2016p913840

horizontal_uncertainty: 13.5701713
depth_errors.uncertainty: 12746.92009

salichon, 2018-10-17:
Hi @fiikennyg 
you can refer to https://quake.ethz.ch/quakeml/docs/REC?action=AttachFile&do=view&target=QuakeML-BED-20130214b.pdf

- horizontalUncertainty
 Circular confidence region, given by single value of horizontal uncertainty.
Unit: m
- depth 
 Unit is meter:
 depth
value162800/value
uncertainty12500/uncertainty
/depth

-lat /lon have their uncertainties in km
FYI @francovm 
Cheers
jerome

closed 2018-10-17 by salichon

calum-chamberlain, 2018-10-17:
It seems like the uncertainties given in the confidenceEllipsoid are in km for GeoNet events, but should be in m for quakeml standard. From the event @fiikennyg stated, the `semi_major_axis_length` is 24.306108. According to QuakeML standards, this should be in m (page 18 of that pdf), but this seems unlikely given that the depth uncertainty (in m) is apparently 12746.92009. It looks like for one of these, the units are wrong, because it doesn't make sense that the longest semi-major axis should be ~1,000 times smaller than the depth uncertainty.
Regarding the lat/lon uncertainties being in km. I'm surprised by this - the quakeML spec states on page 11 that SI units are used, with the exception of angles, where degrees are used. Looking further on at p.32, it looks like the uncertainties in lat-lon might be expected to be in degrees. I would have expected uncertainties of a quantity to have the same units as that quantity.