GeoNet/help #50: Location code inconsistent between station metadata and waveform data

calum-chamberlain, 2018-12-18:
Stumbled across this while trying to remove responses from recent data for FOZ. FOZ broadband consistently uses location-code 10, but the response returned  by obspy FDSN query changes to location-code 11 in 2017.

```python
from obspy import UTCDateTime
from obspy.client.fdsn import Client

client = Client("GEONET")
t1 = UTCDateTime(2017, 10, 10)
t2 = t1 + 30
station, channel, location, network = ("FOZ", "HHZ", "*", "NZ")
st = client.get_waveforms(
   network=network, location=location, station=station, channel=channel, 
   starttime=t1, endtime=t2)
inv = client.get_stations(
   network=network, station=station, channel=channel, starttime=t1, endtime=t2, level="response")
# Note that this returns three channels as per issue #45 
print(st[0].id)
```
NZ.FOZ.10.HHZ
```python
for network in inv:
   for station in network:
      for channel in station:
         print(channel)
```
Channel 'HHZ', Location '10' 
	Time range: 2004-10-13T00:00:00.000000Z - 2015-09-22T18:00:00.000000Z
	Latitude: -43.53, Longitude: 169.82, Elevation: 54.0 m, Local Depth: 0.0 m
	Azimuth: 0.00 degrees from north, clockwise
	Dip: -90.00 degrees down from horizontal
	Channel types: CONTINUOUS, GEOPHYSICAL
	Sampling Rate: 100.00 Hz
	Sensor (Description): Broadband Seismometer (CMG-3ESP)
	Response information available
Channel 'HHZ', Location '11' 
	Time range: 2015-09-23T05:30:00.000000Z - 2017-03-14T21:20:00.000000Z
	Latitude: -43.53, Longitude: 169.82, Elevation: 58.0 m, Local Depth: 0.0 m
	Azimuth: 0.00 degrees from north, clockwise
	Dip: -90.00 degrees down from horizontal
	Channel types: CONTINUOUS, GEOPHYSICAL
	Sampling Rate: 100.00 Hz
	Sensor (Description): Broadband Seismometer (CMG-3ESP)
	Response information available
Channel 'HHZ', Location '11' 
	Time range: 2017-03-14T21:20:01.000000Z - --
	Latitude: -43.53, Longitude: 169.82, Elevation: 58.0 m, Local Depth: 0.0 m
	Azimuth: 0.00 degrees from north, clockwise
	Dip: -90.00 degrees down from horizontal
	Channel types: CONTINUOUS, GEOPHYSICAL
	Sampling Rate: 100.00 Hz
	Sensor (Description): Broadband Seismometer (CMG-3ESP)
	Response information available

---

Note that the location-code changes in 2017 to 11, which does not match the waveform data.

ozym, 2018-12-18:
Hi Calum,

Yes, you have hit us mid-stride on a data correction. We've updated the near-realtime data and are currently working back through the data to fix a location change.  We're about to sync the changes to the archive (and thus available via fdsn) in the next day or so. Just waiting on some exact change times.

The problem stems from a vault being shifted at FOZ in 2015.  The meta-data associated with that change never made it into the official archive, and we only discovered the change recently.

The swap was about 2015/09/23 within a few days (related to concrete setting!). 

I'll update this ticket when I think it should all be correct again.

Mark

calum-chamberlain, 2018-12-18:
Ah! Thanks for those details Mark, good to know!

ozym, 2018-12-19:
This should be corrected and up to date now.   Please reopen this ticket if pain persists ....

closed 2018-12-19 by ozym