GeoNet/help #52: VEXA - Antenna change

pdenys162, 2019-02-27:
The geoNet data lists a change of antenna for VEXA on the following date. 

Model : TRM115000.00
Start of operation : Thu Jan 31 2019

The downloaded rinex files has the antenna change on DOY 038 (7/2/2019). 

Which date is correct?

staylorofford, 2019-02-27:
Hi @pdenys162,

Our **delta** git repository has the antenna swap at `2019-01-30T20:24:30Z`, and I've just checked the RINEX headers between DOY 025-040 and the TRM115000.00 antenna is there on DOY 030, not before, and every day after. It is possible you have an older version of the RINEX files that do not contain the updated headers. I believe the metadata update took a few days to get through due to connection outages at the site. 

While we are still finishing our migration to our new data reception system some marginally supported aspects of our old system are still floating around, like the `/gps` directory on our FTP. If you have been accessing files from here you will not see the updated headers. Files in the `/gnss` directory will have the correct headers as these are updated regularly - the `/gps` folder is only updated once.

pdenys162, 2019-02-27:
OK.  Thanks.
Yes, the the rinex data I have must have been downloaded before the VEXA metadata was updated. 
Thanks

closed 2019-02-27 by pdenys162