GeoNet/help #53: WGTT antenna

pdenys162, 2019-03-21:
The antenna code for the wellington TG GNSS  (WGTT) has changed on DOY 070 2019 from "LEIAT504        LEIS" to " LEIAT504        NONE".  (This appears to have coincided with a receiver change.)

Has the antenna radome been removed?  If not can the antenna code be corrected. 

Paul Denys

staylorofford, 2019-03-21:
Hi @pdenys162, where are you getting this data from? I have just looked over delta and the radome is still LEIS: https://github.com/GeoNet/delta/blob/master/install/radomes.csv

pdenys162, 2019-03-21:
Hi

The data is coming from the old data repository (/gps/rinex rather than /gnss/rinex) as I understand this site is not being supported by the new data repository.

065/WGTT0650.19D.Z:333                 LEIAT504        LEIS                    ANT # / TYPE
066/WGTT0660.19D.Z:333                 LEIAT504        LEIS                    ANT # / TYPE
067/WGTT0670.19D.Z:333                 LEIAT504        LEIS                    ANT # / TYPE
068/WGTT0680.19D.Z:333                 LEIAT504        LEIS                    ANT # / TYPE
069/WGTT0690.19D.Z:333                 LEIAT504        LEIS                    ANT # / TYPE
070/WGTT0700.19D.Z:333                 LEIAT504        NONE                    ANT # / TYPE
071/WGTT0710.19D.Z:333                 LEIAT504        NONE                    ANT # / TYPE
072/WGTT0720.19D.Z:333                 LEIAT504        NONE                    ANT # / TYPE
073/WGTT0730.19D.Z:333                 LEIAT504        NONE                    ANT # / TYPE
074/WGTT0740.19D.Z:333                 LEIAT504        NONE                    ANT # / TYPE
075/WGTT0750.19D.Z:333                 LEIAT504        NONE                    ANT # / TYPE

Regards
Paul

staylorofford, 2019-03-21:
@pdenys162 the site has recently been brought into our new reception system, but the RINEX header metadata during the handover period had not been updated. Thank you for catching this, the RINEX files in the /gnss directory on our FTP should have the correct headers within the next hour.

pdenys162, 2019-03-21:
Thanks for that.
Is it document anywhere which sites are in the old data repository and which are in the new?

Cheers, Paul

staylorofford, 2019-03-21:
In the `/network/marks/` folder of delta, all sites with `GN` network code are not in the new data repository, but we are in the process of taking over WGTT and AUKT, so these will eventually both be in the new data repository.

salichon, 2021-10-05:
should be  done @staylorofford  rifght ? :)

elidana, 2021-10-05:
yes, confirmed the header was corrected at the time.

```
[elidana@hutl16534 tilde]$ curl -O https://data.geonet.org.nz/gnss/rinex/2019/071/wgtt0710.19o.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1618k  100 1618k    0     0   756k      0  0:00:02  0:00:02 --:--:--  756k
[elidana@hutl16534 tilde]$ zgrep "ANT " wgtt0710.19o.gz 
333                 LEIAT504        LEIS                    ANT # / TYPE
```

closed 2021-10-05 by elidana