GeoNet/help #54: Is there any streaming API service?

penguinpowernz, 2019-05-21:
Hi Geonet,

I was wondering if there was any kind of streaming API services available for quake events, perhaps via websockets or NATS or even MQTT?  I wish to get real time earthquakes events and I doubt you would like me hitting the current quakes endpoints every second.  I was hoping you would make the service that backs your push notifications public facing but I have yet to find any information relating to this.

Alternatively I could make a service that hits your quake endpoint every second and make it available to the public so anyone else looking for this type of service won't need to spam your API as well.

Cheers,
Robert

calum-chamberlain, 2019-05-22:
I'm keen to know more about this as well.

danieldooley, 2019-05-23:
Hi there @penguinpowernz and @calum-chamberlain.

Our recommendation to get the earthquake event data is to query our api (documented [here](https://api.geonet.org.nz/)). The rate at which you poll the service should not matter, within reason; a handful of users polling regularly is of no concern.

Cheers

penguinpowernz, 2019-05-26:
Thanks for the reply

closed 2019-05-26 by penguinpowernz