GeoNet/help #57: FDSN_Client problems with get_events

iceseismic, 2019-06-28:
FDSN get_events fails multiple times, without explanation,  only succeeding after trying ca 9 times ( with no code changes) :  So it succeeds, but only after many unsuccessful attempts...

[image]

[image]

francovm, 2019-06-30:
Dear @iceseismic,

I tried the same request using  `obspy.clients.fdsn`, and it works well:
```
from obspy.clients.fdsn import Client as FDSN_Client

client = FDSN_Client("GEONET")

starttime = "2018-11-01 00:00:00.000"
endtime = "2019-07-01 00:00:00.000"
cat = client.get_events(starttime=starttime, endtime=endtime,minlatitude=-39.70,maxlatitude=-37.5
                        ,minlongitude=175.0,maxlongitude=177.5
                        ,minmagnitude=2.5
                        ,mindepth=0.1,maxdepth=32.9)

print(cat)

```

Also using the URL query (will take a tiome to generate a QuakeML)

https://service.geonet.org.nz/fdsnws/event/1/query?starttime=2018-11-01T00:00:00&endtime=2019-07-07T00:00:00&maxlatitude=-37.5&minlongitude=175.0&maxlongitude=177.5&minlatitude=-39.7&minmag=2.5&mindepth=0.1&maxdepth=32.9

This is one is much quicker, since is retrieving a list of event: 

https://service.geonet.org.nz/fdsnws/event/1/query?starttime=2018-11-01T00:00:00&endtime=2019-07-07T00:00:00&maxlatitude=-37.5&minlongitude=175.0&maxlongitude=177.5&minlatitude=-39.7&minmag=2.5&mindepth=0.1&maxdepth=32.9&format=text

Could you detail the error message that your recieving?

salichon, 2019-08-21:
@francovm Close this one ?

iceseismic, 2019-08-21:
Hi,

I still hit timing problems with  get_events when using  Geonet client ; but can bypass sometimes by using the IRIS client

I still think you have a problem

cheers

  *   Stephen

Notice: This email and any attachments are confidential and may not be used, published or redistributed without the prior written consent of the Institute of Geological and Nuclear Sciences Limited (GNS Science). If received in error please destroy and immediately notify GNS Science. Do not copy or disclose the contents.

salichon, 2019-08-21:
Hello @iceseismic , Stephen! 
 Thanks a lot for this
Hmm I m usually not using  python to fetch the quakeml  indeed. 
How long or painful is it ? 
We have to reproduce your queries  i reckon to track this (while in peace time),
Can you send us a problematic request ? 

Thanks  a lot
FYI @francovm 
Cheers 
Jerome

iceseismic, 2019-08-21:
We have to reproduce your queries i reckon to track this (while in peace time),
Can you send us a problematic request ?
Yes – will do ;  just have to replicate one

salichon, 2019-08-22:
Hi @nbalfour 
I reckon this could be useful for our FDSN services bench marking.
I talked briefly to Stephen Bannister about it today.
And we ll get a typical request on which we can test-proof the issue.

Cheers
Jerome

staylorofford, 2022-03-31:
@salichon this possibly relates to maximum query / transfer size. When I have tried to pull large parts of our EQ catalogue I have had errors that are not explicit, i.e., the code fails without any information as to why.

salichon, 2022-07-15:
Kia Ora @iceseismic 
Would you timm allowing check if this issue persists in light of the updates in the fdsn services over time ?

thank you :)

elidana, 2025-04-15:
Closing this.
Still unclear if it is an issue of maximum query / transfer size when time window is too long (seems like might not be for this example, as data can be retrieved with the obspy client) or maybe some timeout on the host where the query was run at the time

closed 2025-04-15 by elidana