GeoNet/help #60: Units of amplitude measurements

calum-chamberlain, 2019-09-06:
I wondered what units the amplitude measurements in GeoNet QuakeML's are in?  QuakeML spec suggests these generic amplitudes should be stored in SI units, but they encourage the used of the unit field but it doesn't look like this is populated, and the amplitudes do not look like SI values.  For example, [this M 2.9](https://www.geonet.org.nz/earthquake/technical/2019p498440) event has ML amplitudes > 1.0, where ML amplitudes are usually displacement (in m for SI units, 1.6m displacement at a decent distance for a 2.9 seems high). e.g. (copied from the QuakeML):
```xml
<amplitude publicID="smi:nz.org.geonet/Amplitude/20190704101041.098734.32827">
  <type>ML</type>
  <genericAmplitude>
    <value>1.620822695</value>
    <lowerUncertainty>0.2354822821</lowerUncertainty>
    <upperUncertainty>0.2354822821</upperUncertainty>
  </genericAmplitude>
  <timeWindow>
    <reference>2019-07-04T10:09:44.39313Z</reference>
    <begin>-14.999999</begin>
    <end>48.81</end>
  </timeWindow>
  <pickID>
    smi:nz.org.geonet/20190704.100934.39-AIC-NZ.WHZ.10.HHZ
  </pickID>
  <waveformID networkCode="NZ" stationCode="WHZ" locationCode="10" channelCode="HH"/>
  <creationInfo>
    <agencyID>WEL(GNS_Primary)</agencyID>
    <author>scamp@kseqp02.geonet.org.nz</author>
    <creationTime>2019-07-04T10:10:41.098811Z</creationTime>
  </creationInfo>
</amplitude>
```

Any chance the units field of the amplitudes can be populated please?

salichon, 2019-09-08:
Hello @calum-chamberlain 
Please refer to https://docs.gempa.de/seiscomp3/current/apps/global_mlv.html#global-mlv
the input units are mm  from seiscomp3 at last  then into the  event quakeml 

(actually i think it s rather mm.s-1 after filtering since measured over the trace of velocimeters) 

... I guess in a ideal world we would be all in  SI (and even no imperial units .. )  ;) 
*but then the usage the  history and the codes..)

Cheers
Jerome

calum-chamberlain, 2019-09-10:
Hi Jerome, thanks for that - any chance the units field of the Amplitude measurement in quakeml can be populated so that users don't have to dig through Seiscomp docs? 

Is the QuakeML generated by Seiscomp, or does that just generate an SC3ML file that you guys convert?

salichon, 2019-09-11:
Kia Ora @calum-chamberlain 
I would deem that the  unit may be deliberately not informed  in generalk since amplitude measurements  may span a fair amount of scale depending of the magnitude/instrument used.
We re are using the provided conversion sheet xslt that you can find under for instance:
 ```https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/libs/xml/0.11/quakeml_1.2__sc3ml_0.11.xsl ```

you may ask  this type of question to Quakeml/seisocmp3   for more details 
See https://quake.ethz.ch/quakeml/
there I think Fabian Euchner woudl provide you with more details thans me :).
Cheers
Jerome

salichon, 2019-09-11:
having had a look at the qml schemes: 

> ***genericAmplitude*** Measured amplitude value for the givenwaveformID. Note that this attribute can describedifferent physical quantities, depending on thetypeandcategoryof the amplitude. These can be, e.g.,displacement, velocity, or a period. If the only amplitude information is a period, it has to specified here,not in theperiod attribute. The latter can be used if the amplitude measurement contains information on, e.g., displacement and an additional period. Since the physical quantity described by this attribute is not fixed, the unit of measurement cannot be defined in advance. However, the quantity has to be specified in SI base units. The enumeration given in attributeunit provides the most likely units that could be needed here. ***For clarity, using the optionalunitattribute is highly encouraged***

this is a request that could be made to seiscomp3 perhaps

calum-chamberlain, 2019-09-11:
Yeah, I read that too. Do you want to suggest it to Seiscomp, or shall I?

salichon, 2019-09-11:
I think it s  always good if it s coming from you ha!  - i ll back your request  up ;)

calum-chamberlain, 2019-09-11:
:+1: Will do.

calum-chamberlain, 2019-09-11:
[Issue opened](https://github.com/SeisComP3/seiscomp3/issues/243) over at Seiscomp - regarding #58  as well, I got feedback from Fabian that the `methodID` field can be used to provide links or DOIs to documentation on how fields are populated/measurements taken.  I think this would be really helpful to provide a consistent and documented catalogue.

salichon, 2019-09-11:
I ve made a question on their forum: https://forum.seiscomp3.org/t/sc3-xml-and-amplitude-units/1518

calum-chamberlain, 2019-09-11:
See comment on [Seiscomp 243](https://github.com/SeisComP3/seiscomp3/issues/243).  Looks like the unit field should be populated if the seiscomp event conversion tool is used.  I guess this comes back here then?

salichon, 2019-09-11:
yeap @calum-chamberlain :) !
The sc3xml to quakeml conversion is made there https://github.com/GeoNet/seiscompml-to-quakeml  - i ll enquire - exploring the built in converter to compare
FYI @nbalfour

salichon, 2019-09-11:
@calum-chamberlain @nbalfour 
here is a  difference between the built in  quakeml converter and the Geonet conversion  
Sc3 built-in: 
`<amplitude publicID="smi:org.gfz-potsdam.de/geofon/Amplitude/20190704101041.098734.32827">
...         <genericAmplitude>
          <value>0.001620822695</value>
...        </genericAmplitude>
        <type>ML</type>
        **<unit>m</unit>** `

Geonet quakeml  conversion 
`      <amplitude publicID="smi:nz.org.geonet/Amplitude/20190704101041.098734.32827">
        <genericAmplitude>
          <value>1.620822695</value>...        </genericAmplitude>
        <timeWindow>
`

So 
-  units are displayed i the sccnv built in case  and  in SI 
-  Geonet s case no units and  not  in SI
 (from https://github.com/GeoNet/seiscompml-to-quakeml)  

I think there could be  room for improvement :)

salichon, 2019-10-06:
@JonoHanson @danieldooley @carolinelittle 
Kia Ora Morena
about quakeml units (and read above discussion):
> "Looks like the unit field should be populated if the seiscomp event conversion tool is used."

I reckon there is an opportunity  here to improve the on-the-fly conversion of the sc3xml to quakeml event files available through the GeoNet FDSN webservice.
and possibly also changing  the way the  GeoNet process is converting the events  sc3xml to quakeml.

I let you check what s the most relevant :)

Cheers
Jerome

salichon, 2020-06-05:
FYI @JonoHanson  just to keep this in mind 
(wrt  FDSNws reference for a paper about magnitude -  refering  to Geonet  station amplitude data source)

salichon, 2021-01-28:
- [ ] Need to update and check the fdsnxml unit output in our current XML conversion system  

FYI @sue-h-gns
..Archeogit_ticketing  for improvement in the fdsn event service

salichon, 2021-02-04:
As a note for later 
2021 in the archive: 
> -	Quakeml  Geonet  provided:  ML Amplitude Unit is millimeter  
> -	Sc3xml (original source): ML Amplitude is millimeter  
No conversion of the amplitude into SI for any of the format 
No Unit description is provided

salichon, 2024-09-04:
- Still pending :(