GeoNet/help #61: NTVZ mismatch between waveform data and station metadata

calum-chamberlain, 2019-10-03:
Hi - just playing with some Taupo data from this year and it looks like there is a mismatch between the station metadata and the available waveforms for station NTVZ around 17/04/2019.  Example using obspy:

```python
from obspy import UTCDateTime
from obspy.clients.fdsn import Client

client = Client("GEONET")
st = client.get_waveforms(
    network="NZ", station="NTVZ", channel="HHZ", location="10",
    starttime=UTCDateTime(2019, 4, 17, 15), endtime=UTCDateTime(2019, 4, 17, 16))
print(st)
```
> 1 Trace(s) in Stream:
NZ.NTVZ.10.HHZ | 2019-04-17T15:00:00.003129Z - 2019-04-17T16:00:00.003129Z | 100.0 Hz, 360001 samples
```python
# Fails with error: no data available
inventory = client.get_stations(
    network="NZ", station="NTVZ", channel="HHZ", location="10",
    starttime=UTCDateTime(2019, 4, 17, 15), endtime=UTCDateTime(2019, 4, 17, 16))
```

If I try and get the station for longer it shows that the station should have stopped on the 17th at 1am:
```python
inventory = client.get_stations( 
     network="NZ", station="NTVZ", channel="HHZ", location="10", 
     starttime=UTCDateTime(2019, 4, 1), endtime=UTCDateTime(2019, 4, 17, 16), 
     level="channel")
print(inventory[0][0][0].end_date)
```
> 2019-04-17T01:10:00.000000Z

Clearly data are available, so it looks like a metadata issue?

salichon, 2019-10-03:
Hi Calum
Yes sometimes the data are there but the metadata starts later (a couple of hours sometimes)
(since often crappy recordings for a couple of hours  or reboot or techs doing other work or tests or …  )

But for this one I reckon it was a  typo setting it 2 days later than the change

This should be corrected
--> I lodged a PR to correct this

Thanks for this
Cheers
Jerome

Notice: This email and any attachments are confidential and may not be used, published or redistributed without the prior written consent of the Institute of Geological and Nuclear Sciences Limited (GNS Science). If received in error please destroy and immediately notify GNS Science. Do not copy or disclose the contents.

calum-chamberlain, 2019-10-03:
Great, thanks!

salichon, 2019-10-03:
https://github.com/GeoNet/delta/pull/627

salichon, 2019-10-03:
Done and merged   @calum-chamberlain  
I let you close after you d check :) 
cheers
jerome

salichon, 2019-12-18:
Date changed and can be seen there: 
`https://service.geonet.org.nz/fdsnws/station/1/query?station=NTVZ&level=channel&format=text `
I m closing this ticket accordingly 
jerome

closed 2019-12-18 by salichon