GeoNet/help #62: New Zephyr 3 Geodetic antennas

pdenys162, 2019-10-16:
Several of the recently replaced Trimble Alloy/ Zephyr 3 Geodetic antenna appear to have an antenna offset:

DUND   height offset ~10 mm
KTIA                            ~ 9 mm
METH                          ~ 6 mm
WARK                          ~ 12 mm
WHNG east offset       ~ 2 mm

See figure attached for WARK.

Are these new antennas defective? 

[image]

staylorofford, 2019-10-16:
Hi @pdenys162,

We are expecting offsets to be introduced by the new antenna model. We produce site ties for each antenna swap which are available at ftp://ftp.geonet.org.nz/gnss/site_ties/ a few days after each swap with some information about what was swapped. Hopefully these can help to remove the offsets you are observing.

pdenys162, 2019-10-16:
Yes, appreciate that there could be offsets
But we do not see it at all sites.  Therefore, the question to be asked is are the antennas faulty.

Regards
Paul Denys

paulagentle, 2019-10-16:
@pdenys162 it might be the antenna that has come off the the site WARK.

I completed a local tie survey in 2012 and during the realisation of the ITRF2014 they discovered a 14mm discrepancy in the vertical and 3.9mm in the north between the space geodesy techniques and the local tie.

Id         DOMES    Soln Id    DOMES   Soln East North Up epoch 
WARK 50243M001 1 7377 50243S001 1 **-0.3 -3.9 14.7** 12:338

Refer to:
https://www.iers.org/SharedDocs/Publikationen/EN/IERS/Publications/tn/TechnNote38/tn38.pdf?__blob=publicationFile&v=4 

The new antenna has been calibrated at GA's IGS calibration facility in Canberra.

elidana, 2019-10-22:
I have informed Sigrun of that as well.
She has processed the data, and also used the tie in data (available from `ftp://ftp.geonet.org.nz/gnss/site_ties`) to better control the offset.

Her results are slightly different from what @pdenys162 is observing, and she wonder if this differences are due to different antenna calibration models used for the processing.

The offsets she records are:

> DUND
> N 3.9 +/- 1.3
> E 2.6 +/- 1.3
> U -1.3 +/- 2.6
> 
> WARK
> N -1.4 +/- 0.8
> E 0.6 +/- 0.7
> U -1.3 +/- 1.7
> 
> These values are completely different from what I would expect when I look at the fits time series.
> Are you guys using the right antenna info? ( I assume so) But also, what antenna phase center model are you guys using for the new antenna?  Old?
> 
> I am concerned about the DUND N and E offset, but the WARK offset is just simply not significant according to the tie data.  So I am a bit baffled by this.
>

elidana, 2019-10-22:
the antenna model used for the GeoNet standard processing (that feed fits time series) is:

```
TRM115000.00    NONE                                        TYPE / SERIAL NO
ROBOT               Geo++ GmbH               5    22-DEC-16 METH / BY / # / DATE
     5.0                                                    DAZI
     0.0  90.0   5.0                                        ZEN1 / ZEN2 / DZEN
     4                                                      # OF FREQUENCIES
IGS14_2035                                                  SINEX CODE
```

pdenys162, 2019-10-22:
I use the antenna PCV values as per the I14.ATX file.

             Before                                        After
DUND TRM57971.00     NONE               TRM115000.00    NONE
WARK TRM55971.00     NONE               TRM115000.00    NONE

Based on the rinex file, there has not been any antenna height change. 

The values I have determined for the position time series offset value is based on the L3 (iono free) solution and not on the single frequency tie values. 

pd

elidana, 2019-10-22:
thanks @pdenys162

elidana, 2019-12-03:
I have reviewed the antenna offsets on all the Zephyr2 to Zephyr3 swaps done so far.
The vertical offset caused by the antenna swaps is here calculated by the difference between the average of 15 days before and after the swap (excluding the day of the swap). 

In the table below, site code, vertical offset and previous antenna type are listed for PositioNZ the sites that had their antenna changed so far (metadata can be found on the `github.com/GeoNet/delta` repository).

site | vertical offset (mm)
--|--
NPLY | -4.08 | TRM57971.00
LEXA | 0.61 | TRM57971.00
MAVL | 1.92 | TRM57971.00
MAHO | 2.60 | TRM57971.00
HAST | 4.49 | TRM57971.00
TRNG | 5.53 | TRM57971.00
CORM | 6.41 | TRM57971.00
WHNG | 6.57 | TRM57971.00
WANG | 6.85 | TRM57971.00
WHKT | 7.07 | TRM55971.00
HIKB | 7.27 | TRM57971.00
GISB | 7.66 | TRM57971.00
PYGR | 7.98 | TRM57971.00
DNVK | 8.97 | TRM57971.00
WRPA | 9.20 | TRM57971.00
AUCK | 9.37 | TRM57971.00
SCTB | 9.99 | TRM57971.00
HAMT | 13.09 | TRM57971.00
METH | 14.28 | TRM57971.00
DUND | 16.04 | TRM57971.00
WARK | 16.84 | TRM55971.00
KTIA | 17.00 | TRM55971.00
CHTI | 17.35 | TRM55971.00

**Horizontal offsets** (North and East component) are smaller and generally within the expected values ( +/- 2 mm), with the exception of **DUND**, that is showing +4 mm offset on the North and  -2.93 mm on the East component.

Graphs for each station (vertical component) are presented below.
Vertical yellow bar is the day of the antenna change, the blue and orange horizontal lines are the average relative position calculated for 15 days before and after the antenna swap, respectively.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2021-09-21 by elidana