GeoNet/help #63: Downloading local EQ waveforms for station TKGZ for plotting record sections

melzhangmelzhang, 2019-11-23:
Hello and good morning,

I am trying to download seismic waveforms from GeoNet stations, in particular from TKGZ, of magnitude 4.5+ during the duration of the station's operation (2013-03-13 until present). I am looking for events originating within the location range 176 to 180 longitude, -36 to -40 latitude.

The goal is to plot the waveforms as a record section, based on event distance from station, in order to spot converted wave arrivals. I have tried obspyDMT for one year of the station's operation:

obspyDMT --datapath event_based_dir --min_date 2014-01-01 --max_date 2015-01-01 --min_mag 5.0 --event_catalog GeoNet --data_source GeoNet --net NZ --cha TKGZ --preset 100 --offset 1800 

This appears to have retrieved a bunch of directories named after event times and IDs, but there are no waveforms and the directories are largely empty (see first screenshot attached). As opposed to when I ran the example from the obspyDMT GitHub:

obspyDMT --datapath event_based_dir --min_date 2014-01-01 --max_date 2015-01-01 --min_mag 7.5 --event_catalog NEIC_USGS --data_source IRIS --net "II" --loc "00" --cha "BHZ" --preset 100 --offset 1800
which was successful and downloaded the waveforms (see second screenshot).

My questions are:

1) How can I get the Hikurangi local events as seismograms, either by refining my first obspyDMT command pasted above, or via other methods?

2) What format are the downloaded seismograms in, and are they compatible and/or could be converted to be plotted in SAC? The goal is to plot record sections of many events at one station (in this case TKGZ) and see the difference in converted phase arrival times.

Thank you and I would much appreciate some advice.
[image]
[image]

salichon, 2019-12-02:
Hello melzhangmelzhang  

`How can I get the Hikurangi local events as seismograms,`
`either by refining my first obspyDMT`
`command pasted above, or via other methods?`

-->>  try the fdsnxml  web service from Geonet https://www.geonet.org.nz/data/tools/FDSN

`What format are the downloaded seismograms in`
`and are they compatible and/or could be converted to be plotted in SAC? `

-->> the data we provide are in miniseed format, use ms2sac program to get  them into the SAC format 

cheers
Regards
ejrome

melzhangmelzhang, 2019-12-02:
Hi Jerry,

Thanks for the help. I used the example with 'curl' to get a sample mseed
download.

curl "
https://service.geonet.org.nz/fdsnws/dataselect/1/query?network=NZ&station=TKGZ&location=10&channel=EH?&starttime=2016-08-31T00:00:00.000&endtime=2016-09-01T11:59:59.000"
-o test.mseed

  % Total    % Received % Xferd  Average Speed   Time    Time     Time
Current

                                 Dload  Upload   Total   Spent    Left
Speed

100 33.2M  100 33.2M    0     0  5219k      0  0:00:06  0:00:06 --:--:--
8004k

I also downloaded and compiled mseed2sac v2.3 but it is giving me an error
message when I read in the mseed file.

(base) Admins-MacBook-Pro:~ melzhang$ ms2sac test.mseed

Header for wrong channel

I get this when I run it from within SAC as well.

(base) Admins-MacBook-Pro:~ melzhang$ sac

 SEISMIC ANALYSIS CODE [11/11/2013 (Version 101.6a)]

 Copyright 1995 Regents of the University of California

SAC> ms2sac ./test.mseed

Header for wrong channel

At first I was trying the command 'mseed2sac' which was not found, then
after using 'ms2sac' it seems it is the correct command but there is a
header or channel issue. Do you know how to fix this?

Mel Zhang
M.S. Student | United Govt. of Grad Students (UGGS) Diversity & Inclusion
Chair
Geological Sciences | University of Colorado, Boulder

calum-chamberlain, 2019-12-02:
@melzhangmelzhang Your question about obspyDMT might be better asked to the [obspyDMT team](https://github.com/kasra-hosseini/obspyDMT/issues).

Not sure what is going on this ms2sac within SAC, but because you already have obspy installed (thanks to obspyDMT) you might find it easier to read the data in using obspy and work on it in Python (or write it out to SAC). The data downloaded by that curl command appear to be valid mseed, and readable by obspy:

```python
from obspy import read

st = read("test.mseed")
for tr in st:
    tr.write("{id}.{starttime}.SAC".format(
        id=tr.id, starttime=tr.stats.starttime), format="SAC")
```

will convert the mseed you download to SAC. I strongly recommend obspy for all your basic seismological data processing! It makes getting data from FDSN servers super easy, checkout the examples GeoNet provides for [using obspy to get their data](https://github.com/GeoNet/fdsn/tree/master/examples)

francovm, 2019-12-02:
HI @melzhangmelzhang, 

As @calum-chamberlain said, I strongly recommend to use Obspy for seismic data processing. We have several tutorials published last year [Python data tutorials](https://github.com/GeoNet/data-tutorials/tree/master/Seismic_Data/Python) to work with the FDSN web services.
In the root of this repo you will find an yml file to set up your enviroments dependencies. 

Cheers,

Franco

salichon, 2019-12-04:
for mseed2sac refer to the documentation 
 https://github.com/iris-edu/mseed2sac/blob/master/doc/mseed2sac.md documentation 
besides ms2sac is a standalone program and not a SAC  command 

Cheers
Jerome

salichon, 2019-12-18:
I  reckon we can close this ticket.
Unless further questions to  re-open it.

closed 2019-12-18 by salichon