GeoNet/help #69: SM instrument trigger level metadata

cvanhoutte, 2020-04-29:
Hi there

I'm wondering whether the trigger level of strong-motion instruments, both in recent times and of days gone by, is available somewhere in the station metadata. I had a squizz at the stationxml files but couldn't find it.

Thanks
Chris

ozym, 2020-04-30:
Currently the data is streamed in real-time and the triggering part is part of the old processing system (which will be replaced at some point).  The configured instrument triggering has two aspects (depending on model), an LTA/STA related to long term/short term averages after various filters have been applied, and a threshold trigger. The two can be used at the same time, the threshold tends to be a backstop at quite a high value.

cvanhoutte, 2020-04-30:
Alrighty, a bit more complex than a threshold then. Exactly the info I was after though, so thanks heaps @ozym

closed 2020-04-30 by cvanhoutte