GeoNet/help #71: Horizontal component orientation of old strong motion FTP data

cvanhoutte, 2020-07-21:
Hi team

I've had a question from an structural engineering end-user about the strong motion data available on the old FTP.

The questions specifically pertains to data from TEPS but apparently other stations may show the same behaviour.

Horizontal components at TEPS changed in 2015 i.e.
```
[Channel 'HN1', Location '20' 
Time range: 2002-03-07T22:20:00.000000Z - 2015-03-10T00:00:00.000000Z
Latitude: -41.29, Longitude: 174.78, Elevation: 111.0 m, Local Depth: 0.0 m
Azimuth: 12.00 degrees from north, clockwise
Dip: 0.00 degrees down from horizontal
Channel types: TRIGGERED, GEOPHYSICAL
Sampling Rate: 200.00 Hz
Sensor (Description): Accelerometer (FBA-ES-T)
Response information available, 

Channel 'HN1', Location '20' 
Time range: 2015-03-10T00:00:05.000000Z - --
Latitude: -41.29, Longitude: 174.78, Elevation: 111.0 m, Local Depth: 0.0 m
Azimuth: 49.00 degrees from north, clockwise
Dip: 0.00 degrees down from horizontal
Channel types: TRIGGERED, GEOPHYSICAL
Sampling Rate: 200.00 Hz
Sensor (Description): Accelerometer (FBA-ES-T)Response information available]
```

For data from the 2013 Cook Strait events, Vol1 to Vol3 files seem to get rotated to NS/EW orientation, but for data from Kaikōura events, no reorientation seems to occur.

What the end user sees is horizontal data from 2013 presented as "N00/N90" but later records as "S41E/S49W". Is this 
- a bug in component orientation metadata that was fixed at some point, or 
- formerly were the horizontal components rotated to NS/EW but this procedure stopped at some point
- another explanation I'm not thinking of?

Thanks
Chris

JonoHanson, 2020-07-27:
@mnaguit and @salichon can you guys help with this?

mnaguit, 2020-07-27:
I have already responded to this via email. But probably @cvanhoutte is looking for further information. I only have a limited understanding of the Fortran code that generates these products. Other insights from @salichon & @ozym would surely help.

Here's my response:
```
From: Muriel Naguit <m.naguit@gns.cri.nz>
Sent: 16 July 2020 17:46
To: Chris Van Houtte <c.vanhoutte@gns.cri.nz>
Subject: RE: TEPS recording station 
 
Hi Chris.
I checked for TEP, there are changes in azimuth, as can be seen from the FDSN station query: 
 
https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=TEPS&level=channel&format=text
 
From 2013 to 2015, the azimuth is as shown in the attached V1 product.
Whereas for a 2016 event, the azimuth from FDSN also corresponds to the V1 product.
 
Note that V1 products are time series corrected only for cross-axis effects & deviations, whereas V2 are fully processed records of time series. I think the orientations in V2 are the same orientations shown in the corresponding V3 response spectra files.
```

cvanhoutte, 2020-08-31:
Hi @ozym or @salichon, do you have any further thoughts on this?
Thanks
Chris

salichon, 2021-10-04:
Instruments has been maintained

closed 2021-10-04 by salichon