GeoNet/help #72: FDSN http resolution failing

calum-chamberlain, 2020-07-22:
Hi all,

I'm not sure if this is temporary or not, but as of this morning I am not able to access the geonet FDSN client via obspy using the http address.  Obspy's resolution of:
```python
from obspy.clients.fdsn import Client

client = Client("GEONET")
```
goes to the http://service.geonet.org.nz address. This returns a `Connection reset by peer error`:
```python
Traceback (most recent call last):
  File "/home/chambeca/miniconda3/envs/eqcorrscan/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/chambeca/miniconda3/envs/eqcorrscan/lib/python3.8/site-packages/obspy/clients/fdsn/client.py", line 1454, in run
    code, data = download_url(
  File "/home/chambeca/miniconda3/envs/eqcorrscan/lib/python3.8/site-packages/obspy/clients/fdsn/client.py", line 1812, in download_url
    buf = io.BytesIO(url_obj.read())
  File "/home/chambeca/miniconda3/envs/eqcorrscan/lib/python3.8/http/client.py", line 467, in read
    s = self._safe_read(self.length)
  File "/home/chambeca/miniconda3/envs/eqcorrscan/lib/python3.8/http/client.py", line 608, in _safe_read
    data = self.fp.read(amt)
  File "/home/chambeca/miniconda3/envs/eqcorrscan/lib/python3.8/socket.py", line 669, in readinto
    return self._sock.recv_into(b)
ConnectionResetError: [Errno 104] Connection reset by peer
---------------------------------------------------------------------------
FDSNException                             Traceback (most recent call last)
<ipython-input-6-35cc17b5ae36> in <module>
----> 1 Client('http://service.geonet.org.nz')

~/miniconda3/envs/eqcorrscan/lib/python3.8/site-packages/obspy/clients/fdsn/client.py in __init__(self, base_url, major_versions, user, password, user_agent, debug, timeout, service_mappings, force_redirect, eida_token, _discover_services)
    261 
    262         if _discover_services:
--> 263             self._discover_services()
    264         else:
    265             self.services = DEFAULT_SERVICES

~/miniconda3/envs/eqcorrscan/lib/python3.8/site-packages/obspy/clients/fdsn/client.py in _discover_services(self)
   1547                    "be due to a temporary service outage or an invalid FDSN "
   1548                    "service address." % self.base_url)
-> 1549             raise FDSNException(msg)
   1550 
   1551         # Cache.

FDSNException: No FDSN services could be discovered at 'http://service.geonet.org.nz'. This could be due to a temporary service outage or an invalid FDSN service address.
```

In contrast, there is no problem using an https version of the same url, e.g.
```python
client = Client("https://service.geonet.org.nz")
```
seems fine.

It would be good to know if this was a temporary thing or whether we should changed the routing in ObsPy to the https address. 

Similarly running a curl request for waveform data via http does not work:
```bash
$ curl "http://service.geonet.org.nz/fdsnws/dataselect/1/query?network=NZ&station=TDHS&location=20&channel=?N?&starttime=2016-09-01T16:37:00.000&endtime=2016-09-01T16:42:00.000" -o test.mseed

curl: (56) Recv failure: Connection reset by peer
```

salichon, 2020-07-22:
Hi Kia Ora  @calum-chamberlain 
Would you try again ?  - it looks fine to me this morning  with the curl command
cheers
jerome

calum-chamberlain, 2020-07-22:
Thanks for getting back to me Jerome!
Unfortunately it still isn't working for me on my home computer, or our VUW servers. I don't know enough about how these requests work to try and debug this sadly.
This may not be a GeoNet issue - I also get the same error for the IRIS FDSN service via http but not via https. I will send some emails and see if anyone else is getting the same errors.

salichon, 2020-07-22:
@calum-chamberlain i tried from diversely located instances and it looks like ok.
Can it be related to your close environment ? - :)

calum-chamberlain, 2020-07-22:
Thanks for checking @salichon - decidedly odd! I'll close this then and continue scratching my head!

closed 2020-07-22 by calum-chamberlain

calum-chamberlain, 2020-07-23:
It looks like a VUW issue, sorry for the spam!  The joys of large organisations IT services not communicating sudden breaking changes!