GeoNet/help #73: missing and/or inconsistent metadata

segburg, 2020-08-07:
I encountered a few missing and/or inconsistent metadata information while measuring seismic waveform amplitudes with scamp (sc3) for a subset of earthquakes in the pre-seiscomp period. It seems that for the cases that I looked at the issues are two-fold:
- missing metadata information for a certain period, and/or
- inconsistent channel codes between the event xml, the metadata and/or waveforms stored on the /geonet/seismic disk.

I have automatically compiled a list (attachment) of the issues found for the subset of larger earthquakes only. Hence this list may not be exhaustive and the time periods indicated are those for which there were earthquakes in my list. 

Example1 (line 1 of the list):
Event file 2103645.xml (2003-08-21T12:12:49.867000) has waveform picks for MTAZ. There are corresponding mseed data on /geonet/seismic but no metadata for this same station before 2007. The output from the fdsn request, https://service.geonet.org.nz/fdsnws/station/1/query?sta=MTAZ&level=channel&format=text, suggests that there is no metadata for MTAZ for 2003:
#Network | Station | Location | Channel | Latitude | Longitude | Elevation | Depth | Azimuth | Dip | SensorDescription | Scale | ScaleFreq | ScaleUnits | SampleRate | StartTime | EndTime
NZ|MTAZ|10|EHZ|-36.786490|174.905600|65.000000|0.000000|0.000000|-90.000000|Short Period Seismometer|74574725.120000|15.000000|m/s|100.000000|2007-06-25T03:00:01|2007-06-25T04:00:00
NZ|MTAZ|10|EHZ|-36.786490|174.905600|65.000000|0.000000|0.000000|-90.000000|Short Period Seismometer|74574725.120000|15.000000|m/s|100.000000|2007-06-26T00:50:00|2012-10-09T01:30:00
NZ|MTAZ|10|EHN|-36.786490|174.905600|65.000000|0.000000|0.000000|0.000000|Short Period Seismometer|74574725.120000|15.000000|m/s|100.000000|2007-06-26T00:50:00|2012-10-09T01:30:00
NZ|MTAZ|10|EHE|-36.786490|174.905600|65.000000|0.000000|90.000000|0.000000|Short Period Seismometer|74574725.120000|15.000000|m/s|100.000000|2007-06-26T00:50:00|2012-10-09T01:30:00

The same is probably true for the lines with True in the 3rd and and 5th columns of the attachment.

Example 2: 

Event file 2103645.xml (2003-08-21T12:12:49.867000) has waveform picks for NGS (seed code NGZ.10.EHZ). The channel code differs from SHZ on the disk /geonet/seismic.
The output of the fdsn request, curl "https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=NGZ&level=channel&format=text", suggests that there is no metadata previous to 2004-03-04 for NGZ:

#Network | Station | Location | Channel | Latitude | Longitude | Elevation | Depth | Azimuth | Dip | SensorDescription | Scale | ScaleFreq | ScaleUnits | SampleRate | StartTime | EndTime
NZ|NGZ|10|EHZ|-39.175584|175.601173|1452.000000|0.000000|0.000000|-90.000000|Short Period Seismometer|74574725.120000|15.000000|m/s|100.000000|2004-03-04T23:00:00|2014-02-25T00:00:00
NZ|NGZ|10|EHN|-39.175584|175.601173|1452.000000|0.000000|0.000000|0.000000|Short Period Seismometer|74574725.120000|15.000000|m/s|100.000000|2004-03-04T23:00:00|2014-02-25T00:00:00
NZ|NGZ|10|EHE|-39.175584|175.601173|1452.000000|0.000000|90.000000|0.000000|Short Period Seismometer|74574725.120000|15.000000|m/s|100.000000|2004-03-04T23:00:00|2014-02-25T00:00:00
NZ|NGZ|10|EHZ|-39.175584|175.601173|1452.000000|0.000000|0.000000|-90.000000|Short Period Seismometer|74574725.120000|15.000000|m/s|100.000000|2014-02-25T00:00:01|
NZ|NGZ|10|EHN|-39.175584|175.601173|1452.000000|0.000000|0.000000|0.000000|Short Period Seismometer|74574725.120000|15.000000|m/s|100.000000|2014-02-25T00:00:01|
NZ|NGZ|10|EHE|-39.175584|175.601173|1452.000000|0.000000|90.000000|0.000000|Short Period Seismometer|74574725.120000|15.00
[missing_metadata.xlsx](https://github.com/GeoNet/help/files/5038901/missing_metadata.xlsx)
0000|m/s|100.000000|2014-02-25T00:00:01|
NZ|NGZ|30|HDF|-39.175584|175.601173|1452.000000|0.000000|0.000000|0.000000|Barometer|4194.304000|15.000000|hPa|100.000000|2014-02-25T03:00:00|

This  is likely the case for all the lines with False in the 3rd and and 5th columns of the attachment.

ozym, 2020-08-07:
Prior to 2007 this data was collected via analogue telemetry, the actual meta data (in terms of amplitude response) is not fully known, the data from this period will pretty much all be 50 Hz (e.g SHZ etc). The sensor response, where known, should be available.

When the digital system was installed the sampling rate was changed to 100Hz and this will reflect the code names (e.g. EHZ) at this point the response is known.

We may need to do some more archaeology to try and join bits together for this early analogue telemetry data.

segburg, 2020-08-11:
So am i understanding right that for those data with channel names SHZ only, the data was collected via analogue telemetry, and arrival times were picked on those streams, but that 100Hz data would be collected later as well?
For example the data in /geonet/seismic/2003/NZ/MTAZ/EHZ.D/ has a sample rate of 100sps. Would the instrument response be known in that case?
What about other cases with channel names HHZ, BNZ, HNZ in the list that i provided?

JonoHanson, 2020-11-08:
For a lot of the analogue sites, the instrument response wasn't fully known or understood, rather the system would have made an estimate based on an amplitude, and that amplitude would have been calibrated with other observations to estimate what the appropriate Wood-Anderson mm would have been. This could then be used for event magnitudes.

salichon, 2021-11-22:
- This could be addressed after further investigation back  in the  time

- Documentaiton and information to be made available through the Delta repository and the metadata products