GeoNet/help #74: Converting miniseed data to strong motion data

SquirrelKnight, 2020-09-14:
Hello,

I am currently downloading a large volume of miniseed data from the GeoNet FDSN server via ObsPy. That aspect has been working quite well. I am, however, curious if there is a way (perhaps a script that has already been written) to convert miniseed data to the strong motion data formats outlined on this page: https://www.geonet.org.nz/data/types/strong_motion

Any advice is greatly appreciated. Thank you for your time.

-Jesse

mnaguit, 2020-09-15:
Hi Jesse. 

If you are keen to use the GeoNet strong motion data products, these are currently available via ftp (ftp://ftp.geonet.org.nz/strong/processed), as described in the link you've cited.

Feel free to contact me via email if you have further queries about the strong motion data products.
My email: m.naguit@gns.cri.nz

Cheers,
Muriel

closed 2021-10-04 by salichon

gsrijayanthi, 2022-02-05:
Can you please let me know, How to convert the broadband seismological data into ground motion data. Are there any tools available ? Could you provide the links to download such tools.

elidana, 2022-02-07:
Hello @gsrijayanthi , 
strong motion data products are available via a 

Strong Motion tool: for more details please go to https://www.geonet.org.nz/data/types/strong_motion
Alternatively, for "bulk data downloads", data are also made available via https://data.geonet.org.nz/seismic-products/strong-motion/

Hope this helps

gsrijayanthi, 2022-02-08:
Thank you very much for the information.