GeoNet/help #75: Slow slip location please - area named GNBK (Greenbank)? where is this

antse78, 2020-10-12:


elidana, 2020-10-12:
Hello @antse78 , thanks for your question.

GNBK is a continuous GNSS site. Its location (latitude and longitude) is available on our delta metadata database.

https://github.com/GeoNet/delta/blob/master/network/marks.csv#L44

Alternatively, you can search for this GNSS site code on our search tool in https://www.geonet.org.nz/data/network/sensor/search
(you will need to select the GNSS/GPS in the sensor type list)

Hope this helps,

Elisabetta

antse78, 2020-10-12:
Awesome thank you Elisabetta

elidana, 2020-10-12:
cheers, glad it helped!

I will "close" this open issue now, but please do not hesitate to reach out again if you have further questions.

closed 2020-10-12 by elidana