GeoNet/help #77: Required FlatFile data for FAS_vertical and PSA_5pc_vertical

sreenathvemula, 2020-12-05:
In the flatfiles provided for the vertical Fourier and PSA data, nearly 540 reports are missing when compared to GeoMean data for the same. Can someone kindly guide me for the calculation of filter cutoff frequencies, Fourier spectra? Also, I tried using Matlab function 'FFT' and tried to match with the Fourier spectrum values provided. The shape I obtained matched with that of the FlatFile, but the values had slight discrepancies. Please help me in the understanding of determining Fourier spectra too.

sreenathvemula, 2020-12-07:
[image]
I tried plotting Fourier spectra for the earthquake on 22/02/2012 on CCCC site

mnaguit, 2020-12-10:
Hi. Regarding vertical vs horizontal, there are different number of vertical vs horizontal data because sometimes the vertical data are of lower quality than horizontal, and need to be rejected. The filter cut-off frequencies are determined by plotting the acceleration FAS and finding where spectrum flattens at high and low frequencies. More detail can be found in the publications cited here: https://github.com/GeoNet/data/blob/main/nzsmd-flatfiles/README.md

closed 2021-10-05 by salichon