GeoNet/help #78: amplitude information missing from the event xml

segburg, 2020-12-16:
I encountered a few event xml files  for which amplitude informations are missing. Contributing station magnitudes are  included, and refer to Mlv amplitude Ids, which are missing.

The publicids with missing amplitude informations are below, but there could be more:
2013p545297.xml
2014p051690.xml
2014p052015.xml
2014p052498.xml
2014p354109.xml
2014p004933.xml
2014p051854.xml
2014p052055.xml
2014p242353.xml
2014p008685.xml
2014p051920.xml
2014p052137.xml
2014p258757.xml

salichon, 2020-12-17:
thanks @segburg 

- [ ] Detail the missing bits of amplitude information
- [ ] Investigate the reason(s)
- [ ] Test if the actual ampliude measurement can be restored
- [ ] Address the event update mechanism in AWS and if updating the event archive with completed information is relevant
FYI @elidana @JonoHanson

salichon, 2020-12-17:
@segburg would you mind to precise the entire  time window of investigation  you worked 
in oredr to  identify if this could be relkated to a unique period of time ? 
thanks a lot

segburg, 2020-12-20:
I was looking at 2012-2020

segburg, 2020-12-21:
There is also this event with an origin id but nothing else: https://service.geonet.org.nz/fdsnws/event/1/query?eventid=2019p431888

segburg, 2021-01-08:
Below is a list of a few more event xmls with MLv station magnitudes but missing MLv amplitudes and no network MLv:
2012p318469
2012p449454
2012p478269
2013p289288
2013p416398
2013p434263
2013p437956
2013p706831
2018p818207
2019p109870

salichon, 2021-01-11:
HNY @segburg !
okay for t 2012-2020
it looks like an an occasional feature. Will  llook at these specific conditions  
and follow up as in  https://github.com/GeoNet/help/issues/78#issuecomment-747723060

salichon, 2021-01-28:
> There is also this event with an origin id but nothing else: https://service.geonet.org.nz/fdsnws/event/1/query?eventid=2019p431888. Quakesearch returns more complete information though:
> 2019p431888,earthquake,2019-06-09T19:10:11.104Z,2019-06-09T19:

@segburg this is looking a different issue but interesting.
I ll  look at it - Check that the sc3xml  is existing as oppoed to the fdsnxml 

- [ ]  Is 2019p431888 informed into our archive ?

salichon, 2021-11-22:
> thanks @segburg
> 
>     * [x]  Detail the missing bits of amplitude information
> 
>     * [x]  Investigate the reason(s)
> 
>     * [x]  Track if it is a systematic feature and list the affected other events
>       -->> occasional cause to detail ( event list as in  [Quake Catalogue - Occasional  amplitude information missing from the event xml #78 (comment)](https://github.com/GeoNet/help/issues/78#issuecomment-756532665) )
> 
>     * [ ]  Test if the actual ampliude measurement can be restored
> 
>     * [ ]  Address the event update mechanism in AWS and if updating the event archive with completed information is relevant
>       FYI @elidana @JonoHanson
> 
> 
> future discussion and work in Geonet/tickets: [GeoNet/tickets#7736](https://github.com/GeoNet/tickets/issues/7736)
> 
> Finalisation here - when understood and done

This still needs to be finalized