GeoNet/help #79: sensor details metadata

caro-gns, 2021-01-22:
This command line: https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=WTMC&level=channel&format=text
states that WTMC has a strong motion sensor from 2010 then an accelerometer from 2017. 
Are these the same sensors? And if so how can I find out if these are Kinemetrics basalt or cusp or etc... please? 
Thank you !

ozym, 2021-01-22:
Two options, either try using `level=response` in the command line (you will also need to remove the `format=text` option) above (rather than `level=channel`).  This will give a full output of the response in XML, which you may need to format  (on linux etc I use `xmllint --format filename.xml` to do this ... e.g. a snippet:

```xml
        <Sensor resourceId="Sensor#CUSP3B:309">
          <Type>Strong Motion Sensor</Type>
          <Description>CUSP 3B</Description>
          <Manufacturer>Canterbury Seismic Instruments</Manufacturer>
          <Model>CUSP3B</Model>
          <SerialNumber>309</SerialNumber>
          <InstallationDate>2005-11-15T01:00:00</InstallationDate>
          <RemovalDate>2010-03-25T22:30:00</RemovalDate>
        </Sensor>
```
or you could check the raw source of the data in the https://github.com/GeoNet/delta repo.  The information in this instance will be buried in  https://github.com/GeoNet/delta/blob/main/install/recorders.csv  . I'd suggest the first option as it's the more formal approach.

The subtlety between a "Strong Motion Sensor" and an "Accelerometer" is another topic ... (but from the outside in a data sense, they are the same thing).

caro-gns, 2021-01-22:
Many thanks  !

Caroline

Note: External sender:

Notice: This email and any attachments are confidential and may not be used, published or redistributed without the prior written consent of the Institute of Geological and Nuclear Sciences Limited (GNS Science). If received in error please destroy and immediately notify GNS Science. Do not copy or disclose the contents.

elidana, 2021-01-25:
Hi @caro-gns ,

thanks to @ozym's reply I think this ticket can be now closed. Please re-open it if you think more details are needed.

closed 2021-01-25 by elidana